exelban/stats #108: Long disk name doesn't fit

jaykovski, 2020-10-11:
Long disk name doesn't fit in popup, Big Sur have a weird long name for a disk
[image]

Proposed solution: truncate long names with "..."

macOS Big Sur Beta 9
27" iMac 2019 (iMac19,1)

exelban, 2020-10-11:
Hmm, yeap it must be fixed. 

Strange name, this disk must be invisible in theory. It's a real disk or just some "recovery" disk or disk with update?

jaykovski, 2020-10-11:
> Hmm, yeap it must be fixed.
> 
> Strange name, this disk must be invisible in theory. It's a real disk or just some "recovery" disk or disk with update?

This is the main and only disk, must have been renamed by OS itself.

exelban, 2020-10-11:
Does it have a normal name in finder or disk utility? 
I try to understand if this is some bug in the app or something with macOS?

jaykovski, 2020-10-11:
> Does it have a normal name in finder or disk utility?
> I try to understand if this is some bug in the app or something with macOS?

It does have a normal name in Finder.

[image]

exelban, 2020-10-11:
Hmm, it looks like something changed in the new version of macOS. 
The app detects the wrong disk name (it must take from the container above). I will fix this bug when the Big Sur will release (or at least will be the release date).

exelban, 2020-10-11:
Thanks for help

exelban, 2020-10-18:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.3.8) release.

closed 2020-10-18 by exelban