exelban/stats #109: Nvidia GPU not detected on 15" MBP

wintercharm, 2020-10-11:
[image]

Basically exactly what it says on the tin -- the Nvidia GPU is not being detected. No stats related to it will show. Wonder if this is an Nvidia GPU-only issue, or an automatic graphics switching problem. 

When toggling to dGPU only, there is no difference, the sensors keep looking at the iGPU, which is reporting zero activity (since it's turned off).

exelban, 2020-10-12:
Hi.

It looks like the app cannot properly read all specs of Nvidia GPU. The only way to fix it runs a debug version of the app. It allows to debug where is the problem. If you have some time and want help I can send you a debug version.

exelban, 2020-10-18:
Please reopen this issue, If someone would like to help with this.

Spazholio, 2020-11-03:
I'm having the exact same issue.  Let me know what I can do to help track it down.

exelban, 2020-11-05:
@Spazholio Hi. Thanks.
Post please the result of `ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt`. It will help to debug why gpu is not visible.

Spazholio, 2020-11-05:
@exelban Sure thing:
`    {   
      "IOClass" = "nvAccelerator"
      "MetalStatisticsName" = "GeForceStatistics"
      "IOVARendererSubID" = <03000000>
      "IODVDBundleName" = "GeForceVADriver"
      "NVDA,Enable-A2R10G10B10Format" = <01000000>
      "NVDA,mm-version" = 4 
      "IOReportLegendPublic" = Yes 
      "IOGLBundleName" = "GeForceGLDriver"
      "IOProviderClass" = "IOService"
      "AGCInfo" = {"fBusyCount"=0,"poweredOffByAGC"=No,"fSubmissionsSinceLastCheck"=61,"fLastSubmissionPID"=78903}
      "MetalPluginName" = "GeForceMTLDriver"
      "IOProbeScore" = 100 
      "SurfaceList" = ()
      "IOSourceVersion" = "14.0.32"
      "IOPropertyMatch" = ({"NVArch"="GF100","nubType"="NVDAgl","unmatched"="yes","NVDAType"="Official"},{"NVArch"="GK100","nubType"="NVDAgl","unmatched"="yes","NVDAType"="Official"})
      "PerformanceStatisticsAccum" = {"orphanedReusableVidMemoryHitRate"=46,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=57,"dataBufferCount"=0,"oolTexturePageInBytes"=0,"textureCount"=4827,"iosurfaceTextureCreationCount"=0,"surfaceReadLockIdleWaitTime"=0,"freeDat#
      "MetalPluginClassName" = "NVMTLDevice"
      "IOAccelRevision" = 2 
      "CFBundleIdentifierKernel" = "com.apple.GeForce"
      "IOAccelDisplayPipeCapabilities" = {"DisplayPipeSupported"=Yes,"TransactionsSupported"=Yes}
      "InternalStatisticsAccm" = {}
      "IOMatchCategory" = "IOAccelerator"
      "CFBundleIdentifier" = "com.apple.GeForce"
      "IOVABundleName" = "GeForceVADriver"
      "PerformanceStatistics" = {"orphanedReusableVidMemoryHitRate"=46,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=57,"dataBufferCount"=0,"oolTexturePageInBytes"=0,"textureCount"=4827,"iosurfaceTextureCreationCount"=2,"surfaceReadLockIdleWaitTime"=0,"freeDataBuff#
      "InternalStatistics" = {}
      "IOCFPlugInTypes" = {"ACCF0000-0000-0000-0000-000a2789904e"="IOAccelerator2D.plugin"}
      "NVDA,NVPresentment-version" = <11000000>
      "IOGeneralInterest" = "IOCommand is not serializable"
      "IOReportLegend" = ({"IOReportChannels"=((4000,6442450945,"Overall GPU Busy"),(4001,6442450945,"Overall GPU total time"),(0,6442450945,"Command Buffer Bytes"),(1,6442450945,"GL Command Buffer Bytes"),(2,6442450945,"CL Command Buffer Bytes"),(3,6442450945,"2D Command Buffer Bytes"),(4,6442450945,"Video#
      "AccelCaps" = 15
      "IOVARendererID" = <08000401>
    }   

+-o IntelAccelerator  <class IntelAccelerator, id 0x10000041e, registered, matched, active, busy 0 (60854 ms), retain 211>
    {
      "IODVDBundleName" = "AppleIntelHD5000GraphicsVADriver"
      "InternalStatisticsAccm" = {}
      "EnableBlitLib" = 1
      "InternalStatistics" = {}
      "IOGVABGRAEnc" = "Gen75"
      "AGCInfo" = {"fBusyCount"=0,"poweredOffByAGC"=No,"fSubmissionsSinceLastCheck"=0,"fLastSubmissionPID"=79197}
      "HwCtxCacheUpdate" = 1
      "MetalPluginName" = "AppleIntelHD5000GraphicsMTLDriver"
      "PerformanceStatisticsAccum" = {"orphanedReusableVidMemoryHitRate"=0,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=62,"dataBufferCount"=0,"oolTexturePageInBytes"=0,"textureCount"=1856,"iosurfaceTextureCreationCount"=0,"surfaceReadLockIdleWaitTime"=0,"freeData#
      "PerformanceStatistics" = {"orphanedReusableVidMemoryHitRate"=0,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=62,"dataBufferCount"=0,"oolTexturePageInBytes"=0,"textureCount"=1856,"Device Utilization % at cur p-state"=0,"iosurfaceTextureCreationCount"=0,"freeD#
      "IOGLBundleName" = "AppleIntelHD5000GraphicsGLDriver"
      "IOPCIClassMatch" = "0x03000000&0xff000000"
      "IOSourceVersion" = "0.0.0.0.0"
      "TelemetryDisable" = 0
      "IOPCIPrimaryMatch" = "0x0d268086 0x0a268086 0x0a2e8086 0x0d228086 0x04128086"
      "IOGVAH264EncodeCapabilities" = {"VTQualityRating"=50,"VTRating"=400}
      "VRAM,totalMB" = 1536
      "PanicOnGPUHang" = 0
      "MetalCoalesce" = 1
      "ForceDisableEDRAM" = 0
      "CompactVRAM" = 1
      "CFBundleIdentifier" = "com.apple.driver.AppleIntelHD5000Graphics"
      "IOVARendererID" = 17301508
      "IOProviderClass" = "IOPCIDevice"
      "IOCFPlugInTypes" = {"ACCF0000-0000-0000-0000-000a2789904e"="IOAccelerator2D.plugin"}
      "IOGVACodec" = "Gen75"
      "IOReportLegendPublic" = Yes
      "IOAccelRevision" = 2
      "IOClass" = "IntelAccelerator"
      "CFBundleIdentifierKernel" = "com.apple.driver.AppleIntelHD5000Graphics"
      "IOGVAScaler" = "Gen75"
      "IOReportLegend" = ({"IOReportChannels"=((4000,6442450945,"Overall GPU Busy"),(4001,6442450945,"Overall GPU total time"),(0,6442450945,"Command Buffer Bytes"),(1,6442450945,"GL Command Buffer Bytes"),(2,6442450945,"CL Command Buffer Bytes"),(3,6442450945,"2D Command Buffer Bytes"),(4,6442450945,"Video#
      "IOMatchCategory" = "IOAccelerator"
      "AccelCaps" = 15
      "IOProbeScore" = 1000
      "IOAccelDisplayPipeCapabilities" = {"DisplayPipeSupported"=Yes,"TransactionsSupported"=Yes}
      "IOGeneralInterest" = "IOCommand is not serializable"
      "SurfaceList" = ()
    }
`

exelban, 2020-11-05:
@Spazholio thanks. Strange. It must be visible.

exelban, 2020-11-05:
@Spazholio could I ask you to run this build. It will write some logs to file `log.txt`. The file can be found in `Documents`. Could you please send me this file or post an output?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5495093/Stats.dmg.zip)

Spazholio, 2020-11-05:
Yep, here's the file:

[log.txt](https://github.com/exelban/stats/files/5495319/log.txt)

exelban, 2020-11-05:
Hmm, it looks like it cannot match `IOPCIDevice` and `IOAccelerator`.
NVidia driver does not contain `IOPCIPrimaryMatch` param. It's necessary to match the gpu adapter and pci device.

exelban, 2020-11-05:
@Spazholio check please this one. I suppose it will show the nvidia gpu. But I cannot obtain a model name. So it must be just `NVidia`.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5495789/Stats.dmg.zip)

Spazholio, 2020-11-05:
Looks good.  It shows up as "Unknown":

[image]

Unsure why there are 2 graphs per card though, unless that's intentional.  The internal card really shouldn't be showing anything, as I have it disabled and am using the Nvidia one exclusively.

exelban, 2020-11-05:
@Spazholio nice. I will fix the name. It will be `Nvidia Graphics`. 

What do you mean by 2 graphs per card? On each card, there are 2 charts: temperature and usage.

Spazholio, 2020-11-05:
Ah!  Since it had never worked for me, I didn't know what I was supposed to be seeing.  My mistake.  =)

Although, what's weird to me is that I'm using the Nvidia card, yet that temperature is reading 32 degrees whereas the Iris Pro is reading a lot higher.  I have "Automatic graphics switching" turned off, and I use an app to make sure that I am always utilizing the Nvidia.  So the fact that it thinks its using the integrated Iris Pro is puzzling.

exelban, 2020-11-05:
32 F it's 0 C. It just means that this GPU does not provide the temperature. So I cannot display it.

Spazholio, 2020-11-05:
Damn it, Nvidia.  Get your crap together.

Ah well - thanks for the fixes you were able to put in place.

exelban, 2020-11-05:
Yeap, it's not only Nvidia. There is a similar problem with AMD.
this fix will be available in the next release. But you can use this build till that.

bouscram, 2020-11-07:
HI. I had a similar problem with my end-2012 iMac, which has a hidden Intel GPU and a nVidia GT650. Until the custom build you refer to above, the nVidia GPU was not available in the GPU list of the GPU panel (while correctly shown in the main panel).

The custom build brings the same correction than explained above :-)

[image]

bouscram, 2020-11-07:
A quick question: in an answer above, you say that the name will be "NVidia Graphics" in the corrected official version. But if you already know the full name, as shown in my screenshot, couldn't you display it?

exelban, 2020-11-07:
I could, but the new build must be compiled and signed. Tomorrow I will release a new version with all GPUs and Fans fixes.

bouscram, 2020-11-07:
For next one then...  ;-) Thanks for you work

exelban, 2020-11-08:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.2) release.

closed 2020-11-08 by exelban