exelban/stats #111: Feature request: Monitoring two GPUs at the same time in the menu bar.

hazarek, 2020-10-12:
Monitoring two GPUs at the same time in the menu bar. You need to click on the icon to see both of them. It would be nice if two graphics cards were displayed simultaneously in the menu bar.

[image]

exelban, 2020-10-12:
Hi. 

I describe in #93 why it's not possible now. But in the future, it can be added.

hazarek, 2020-10-12:
hi @exelban. Can you put a label to find which graphics processor is in use.
for example; `GPU 1`, `GPU 2` or 
`eGPU`= external
`iGPU`= integrated
`dGPU` = discrete

[image]

DevLiuSir, 2020-10-19:
**Can you consider adding more detailed device information for monitoring, such as Model Number: A2141, etc.**

exelban, 2020-10-19:
Yeap, but I will need some design for this

hazarek, 2020-10-19:
> Yeap, but I will need some design for this.
> Also will be better if you do a separate issue for this.

Did you answer my question or @DevLiuSir's question, please use the `quote reply`

exelban, 2020-10-19:
> > Yeap, but I will need some design for this.
> 
> > Also will be better if you do a separate issue for this.
> 
> 
> 
> Did you answer my question or @DevLiuSir's question, please use the `quote reply`

😄
Sorry for this. It was answer on @DevLiuSir question. 

Now the answer on your propose.
I must to take a looks if it's possible to change the name value on fly in this widget. But I think yes, it can be added.

DevLiuSir, 2020-10-19:
> Yeap, but I will need some design for this.
> Also will be better if you do a separate issue for this.

I mean adding more device information to the settings-stats.

exelban, 2020-10-19:
> > Yeap, but I will need some design for this.
> > Also will be better if you do a separate issue for this.
> 
> I mean adding more device information to the settings-stats.

Sorry, but how it's related to multi gpus monitoring? If you want this information to be added, please open an issue for this.

exelban, 2020-12-20:
Hi. I implement an option to show GPU type in mini widget as you propose:
eGPU= external
iGPU= integrated
dGPU = discrete

It's available in the [new](https://github.com/exelban/stats/releases/tag/v2.4.9) release.

exelban, 2020-12-20:
For now, I don't see an option to make monitoring for 2 GPUs at the same time. But I suppose it will be available in the future.

closed 2020-12-20 by exelban

hazarek, 2020-12-20:
@exelban thanks