exelban/stats #113: Feature request: Network update interval

bdkbdk123, 2020-10-13:
It would be really useful to let users set update interval on network stats, just like in CPU, GPU, RAM, etc.

exelban, 2020-10-13:
Hi.

What do you mean?
Do you want to see the average transfer in the last 5 seconds?

bdkbdk123, 2020-10-13:
Not exactly, I would like to be able to set update interval for network stats.
Just like for CPU here in red frame:
[image]

Network settings - no update interval setting (i think it defaults to 1 second):
[image]

exelban, 2020-10-13:
Yes. But network reader works in a little bit different way.
It read the transferred data in some time interval. Let's say now is sec 0. I can read the transferred data after some amount of time. Let's say 5 seconds. So I will receive the transferred data in the last 5 seconds. 

There is only one reason why I add an update interval option. To reduce the number of reads in the Disk module. Which allows reducing CPU usage (each read takes some CPU).

And it was easy to add it to CPU, Memory, and Sensors modules. Because they work in the same way. Battery and Network work in a different way. And adding this option will only increase the CPU usage when reading the value.

exelban, 2020-10-13:
In the CPU, GPU, RAM, DISK, Sensors modules readers can read the value in this moment of time. So it's possible to runs with different time intervals. The battery and network modules it's a time-based value.

bdkbdk123, 2020-10-13:
Thanks for detailed answer and awesome app.
The reason I propose this feature is because it is slightly distracting to see networks usage change every second. Maybe it would be possible to update the network usage in nav bar (just presentation) on users defined interval?

Great job nonetheless!

exelban, 2020-10-13:
Hmm.

I could add the option `Update value interval`. In this case, only the value in the menu bar will update in this interval. Reader will read the value normally (every second), but the value will update every 5 seconds (for example).

But also, it will be a little bit confusing. But I understand that someone doesn't need to see actual network transfer. 

Thanks.

exelban, 2020-10-19:
Hi.

I'm sorry but I will not add this option. This option's a little bit confusing for me. From my point, network transfer must be visible live.

closed 2020-10-19 by exelban

gingerbeardman, 2021-05-14:
It's a shame that this will not be added, given multiple requests it seems some users do want it and do not mind if network update is lagged.

I would argue that even 1 second update is not live.

Though I can see both opinions.

B0R9F3D9, 2024-12-05:
> Hi.
> 
> I'm sorry but I will not add this option. This option's a little bit confusing for me. From my point, network transfer must be visible live.

I think you should reduce the time of updating the information about the internet channel or update this data by pressing the refresh button in the popup. Sometimes the router switches me from 5GHz to 2.4GHz and I can't see it through Stats, it helps only with the option key to click on wifi, there I can see the current mode