exelban/stats #114: Removable drive shown even when not enabled

igorkulman, 2020-10-14:
**Describe the bug**
I have a removable (USB) drive connected to the computer called "Backup". In disk settings I have "Show removable drives" disabled but it is shown

[image]

(Also it probably should be shown just as one "Backup", not 2 separate volumes as all APFS drives have a "- Data" volume, for example the internal "Catalina" drive is shown just once, not as "Catalina" and "Catalina - Data")

macOS clearly thinks the drive is removable

[image]

**Details:**
 - MacOS 10.15.7
 - Application version 2.3.7

exelban, 2020-10-14:
Hi. Thanks for the report. 

Yeap, I noticed this bug with ` - Data` when APFS. It's not present in the main disk, because I detect the main disk and trying to find the correct one container. I must do it with all APFS disks, not only the main one. It must be fixed.

The ejected disk. I read the field from macOS which says if the disk is removable or not. And I suppose macOS report the wrong value here. It works perfectly with usb drives and dmg. I don't have any external drives to check this theory. But I will try to debug it.

igorkulman, 2020-10-14:
What commands do you use for the detection? I can run them locally and report back the results.

igorkulman, 2020-10-14:
Looking at the source code and the data, you are right, the drive returns `DAMediaRemovable=0` and `DAMediaEjectable=0`. 

The only notable difference in data compared to an internal drive is `DADeviceProtocol=USB`

exelban, 2020-10-14:
Thanks.

Check if external disk present in the output of this command: `system_profiler SPSerialATADataType`.
If not, it must be there: `system_profiler SPUSBDataType`.

Don't paste all output.

exelban, 2020-10-14:
Maybe this disk is `Detachable Drive`.

igorkulman, 2020-10-14:
> Thanks.
> 
> Check if external disk present in the output of this command: `system_profiler SPSerialATADataType`.
> If not, it must be there: `system_profiler SPUSBDataType`.
> 
> Don't paste all output. Just the external drive.

The disk is not present in `system_profiler SPSerialATADataType`, it only shows SATA connected internal drives.

The disk is shown in `system_profiler SPUSBDataType` attached to an USB controller.

```
USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCIPPT
      PCI Device ID: 0x1e31
      PCI Revision ID: 0x0004
      PCI Vendor ID: 0x8086

        Portable SSD T5:

          Product ID: 0x61f5
          Vendor ID: 0x04e8  (Samsung Electronics Co., Ltd.)
          Version: 1.00
          Serial Number: 1234567CFAA2
          Speed: Up to 5 Gb/s
          Manufacturer: Samsung
          Location ID: 0x14500000 / 3
          Current Available (mA): 900
          Current Required (mA): 896
          Extra Operating Current (mA): 0
          Media:
            Portable SSD T5:
              Capacity: 500,11 GB (500 107 862 016 bytes)
              Removable Media: No
              BSD Name: disk4
              Logical Unit: 0
              Partition Map Type: GPT (GUID Partition Table)
              S.M.A.R.T. status: Verified
              USB Interface: 0
              Volumes:
                EFI:
                  Capacity: 209,7 MB (209 715 200 bytes)
                  File System: MS-DOS FAT32
                  BSD Name: disk4s1
                  Content: EFI
                  Volume UUID: 0E239BC6-F960-3107-89CF-1C97F78BB46B
                disk4s2:
                  Capacity: 499,11 GB (499 113 885 696 bytes)
                  BSD Name: disk4s2
                  Content: Apple_APFS
```

exelban, 2020-10-14:
Yeap, so it's said that this drive is not removable. The only way to detect it's a USB interface. But I'm not sure if it's a good idea.

exelban, 2020-11-15:
Unfortunately, I did not find any good solution to fix this. The ` - Data` disk will be removed when I figured out how I can recreate this behavior (I don't have any external drive now).

closed 2020-11-15 by exelban