exelban/stats #115: Percentages seem to conflict with Activity Monitor

aubreypwd, 2020-10-14:
One process at least is using `~93%` in Activity monitor while my icon reports `~28%-35%`? Why are these numbers inconsistent?

[image]

exelban, 2020-10-14:
Because in Activity Monitor you see a unix style usage.

closed 2020-10-15 by exelban

aubreypwd, 2020-10-19:
[image]

If I'm groking this right, these numbers are more related?

exelban, 2020-10-19:
Yeap, it's a sum of these values.

aubreypwd, 2020-10-21:
Not, I also confirmed this:

[image]

I guess it's rounding up a bit...

exelban, 2020-10-21:
Nope. it depends on which moment the read was done. Even if the difference between the moment when one app (stats) read the value and another one (IPG) is 10ms. The value can be different.

aubreypwd, 2020-10-26:
Thanks for all your hard work and your clarifications! Great product!