exelban/stats #118: [BUG] Wrong network speed

xfzka, 2020-10-16:
**Describe the bug**
Wrong network speed, may be this app is show total network traffic?

[image]

[image]

**Details:**
 - MacOS 10.15.7 
 - Application version 2.3.7 (2.3.6 also has this bug, I am use this app start  from 2.3.6 version)

exelban, 2020-10-16:
Hi.

Did you use the internal network adapter or some thunderbolt adapter?

xfzka, 2020-10-17:
I use internal network adapter and wifi ,  internal network adapter first.

exelban, 2020-10-19:
So it's an internal wifi adapter?

It's strange behavior. I just read the value from the system. There is no place where I can do the wrong calculation or something.

HartwigHarder, 2020-10-20:
I do have the same issue, using a USB dock, but the network speed in the system overview is reasonable. 
Stats does show the correct adapter and network usage from the top process list when opening the stats menu.

exelban, 2020-10-20:
Hi.

I know that there is a problem with external interfaces. 
It looks like I need to find some new way to read the transferred data size.

exelban, 2020-11-29:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.5) release.

closed 2020-11-29 by exelban