exelban/stats #121: GPU in menu bar not showing temperature of second GPU

davezzz07, 2020-10-20:
Hi there,

First off, the app is very useful and well designed. Congrats! 

I've been using stats for the past month and I have noticed that the GPU menu on the menu bar is not showing the temperature of one of the GPUs that my MacBook Pro 16 has (AMD Radeon Pro 5500M) when it is activated when an external monitor is connected. I see a spike in temperature of the build-in GPU (Intel UHD Graphics 630) of almost 30 degrees but the AMD Radeon temperature remains showing zero. Here are a couple of screenshots to show you the issue.

[image]
[image]

Thank you.

Best.

exelban, 2020-10-20:
Hi. Thanks for reporting.

Maybe this GPU reporting the temperature with a different key.
Could you maybe post the output of `ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt`. It will help to debug why temperature is not available.

davezzz07, 2020-10-21:
You're welcome.
Sure. You can find it attached.
[output.txt](https://github.com/exelban/stats/files/5416360/output.txt)

exelban, 2020-10-21:
Thanks. I will take a look later today.

davezzz07, 2020-10-21:
You're welcome. Take care.

exelban, 2020-10-21:
One more question, did the card was enabled when you run this command?
Because it looks like this output does not have the GPU utilization (amd card).

exelban, 2020-10-22:
Thanks. I take a look. 
The driver reports a 0 value. `"Temperature(C)"=0`. Unfortunately, I can't do anything to fix this.

davezzz07, 2020-10-22:
Sure. Thank you.

exelban, 2020-10-22:
I hope Apple or AMD will fix it. I suppose it's related to drivers and can be fixed in future updates.

closed 2020-10-22 by exelban