exelban/stats #124: Clicking outside

0zguner, 2020-10-24:
**Describe the bug**
Clicking outside does not close the window

**Details:**
 - Device: [e.g. Macbook Pro 2016]
 - MacOS [e.g. 10.15.7]
 - Application version [e.g. 2.3.8]

exelban, 2020-10-24:
Hi.

What window does not close?

0zguner, 2020-10-24:
Like this one
[image]

exelban, 2020-10-24:
Hmm, strange. So its close only with the same method as opens? (by click ob menu bar icon)

0zguner, 2020-10-24:
To be more clear, its hidden when another app is focused
I've also disabled desktop with this, maybe its causing this
```defaults write com.apple.finder CreateDesktop false```

exelban, 2020-10-24:
Sorry, but I do not understand. There is a way to close it? By clicking on the menu bar icon?
Or it always visible and cannot be closed after opening?

closed 2020-10-24 by 0zguner