exelban/stats #125: Enabling different sensors values, change the background to white

jcconca, 2020-10-26:
In some cases, enabling sensors to be visualised, like "Voltage", make the background of the sensor section white, and imposible to read the value.

See the below screenshots, to more information:
[image]

After selecting values in voltage:
[image]

exelban, 2020-10-30:
Hi.

Thanks for reporting. Unfortunately, I do not know why it's happening. There is no any colorizing in code. 
Maybe someone could help with this. It's happening randomly on some systems. And I cannot reproduce this behavior.

exelban, 2020-11-08:
Hi.
I think I know where is the problem. But it requires rewriting the widgets. I already started to do this. Must be ready till the next release.

entee55, 2020-11-08:
I think this issue is also with dark mode. For example in dark mode the graph background colors change to white. When you have the white option selected in colors, then it is the same as the background color.
[image]
[image]

exelban, 2020-11-08:
> I think this issue is also with dark mode. For example in dark mode the graph background colors change to white. When you have the white option selected in colors, then it is the same as the background color.
> [image]
> [image]

If the problem is not related to the issue please open a separate issue. 
In this case, I don't see any problems at all. You chose a white color option. And what you expect, that it will change to the black?

PS. If you want to have monochrome colors, there is a special one. Called monochrome.

entee55, 2020-11-08:
I understand your point, I thought the same.  I meant, that in this case maybe the problem is with the background color is changing due to dark mode. I just wanted to help this issue. For me it is working as expected.

exelban, 2020-11-11:
@entee55 hi. Could I ask you to check this build?
To fix this issue, I must rewrite the sensors widget. And as you remember the last time it crashes the app(
I just don't want to repeat that situation (when the app crashes).

[Stats.dmg.zip](https://github.com/exelban/stats/files/5523339/Stats.dmg.zip)

exelban, 2020-11-15:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.3) version.

closed 2020-11-15 by exelban