exelban/stats #128: Some of the sensors are not shown

wangzhaoming, 2020-10-28:
**Describe the bug**
I noticed that the code defined a SMC key list in the `SensorsList` , but some of them are not shown in the widget.

After doing some debug, I found some keys the SMC returned are not fully matched with the pre-defined keys.
Part of the keys from SMC:
  - 752 : "TC0E"
  - 753 : "TC0F"
  - 754 : "TC0P"
  - 755 : "TC1C"
  - 756 : "TC2C"
  - 757 : "TC3C"
  - 758 : "TC4C"
  - 759 : "TCBC"
  - 760 : "TCGC"
  - 761 : "TCIC"
  - 762 : "TCSA"
  - 763 : "TCXC"

and here is the defined keys:
   ```
Sensor_t(key: "TC0c", name: "CPU core 1", group: SensorGroup.CPU.rawValue, type: SensorType.Temperature.rawValue),
Sensor_t(key: "TC1c", name: "CPU core 2", group: SensorGroup.CPU.rawValue, type: SensorType.Temperature.rawValue),
Sensor_t(key: "TC2c", name: "CPU core 3", group: SensorGroup.CPU.rawValue, type: SensorType.Temperature.rawValue),
Sensor_t(key: "TC3c", name: "CPU core 4", group: SensorGroup.CPU.rawValue, type: SensorType.Temperature.rawValue),
Sensor_t(key: "TC4c", name: "CPU core 5", group: SensorGroup.CPU.rawValue, type: SensorType.Temperature.rawValue),
```

The last letter `c` is lower case while the SMC returned with upper case.
In addition, my cpu has 4 cores and the SMC returned with TC1C ~ TC4C. The index starts from 1. 

I guess that the case of letters and the index are related to the device model or MacOS version. Some different devices may return uppercased `TC1C` or the index of the cpu core will start from 0?

Unfortunately, I didn't found any exact or offical descriptions of the SMC keys. 

**Details:**
 - Device: MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports)
 - MacOS [e.g. 10.15.7]
 - Application version [2.3.8]

exelban, 2020-10-28:
Hi.

Thanks for debugging this issue. These missing sensors `TC0E`, `TC0F` could be added, but you must describe what they indicate. 

Now core sensors. Yes, I know that some version of macs reports with different `c` cases. And it can be fixed by adding `TC0C ~ TC4C` to the list.

Unfortunately, there is no list of sensors with descriptions at all. You could only find some sensors on forums or in some SMC kexts. It makes collecting the list of sensors a little bit hard.

wangzhaoming, 2020-10-28:
Sensors named `TC0E`, `TC0F` are already included in the code and I didn't list them above. Just leave them :grin: . But the names `CPU 1` & `CPU 2` are somewhat ambiguous.
[image]

Yes, the biggest problem is how to find the descriptions of sensors. May someone can provide it 🔍 .

exelban, 2020-10-28:
The problem also that the same name of keys is different sensors in the different macs.

exelban, 2020-11-01:
I add missing sensors in the [last](https://github.com/exelban/stats/releases/tag/v2.4.0) version.

closed 2020-11-01 by exelban