exelban/stats #130: Support other platforms

hmsjy2017, 2020-10-30:
`Stats` is a good software I have never seen. I want to use it in other platforms such like Windows and Linux.

exelban, 2020-10-30:
Hi. Thanks.

I'm not sure that is possible. As I remember windows do not have good support of Swift. Also, do not have a menu bar where the widget can be placed (correct me if now something changed).

With Linux the same problem. Swift is supporting, but I'm not sure that there is a place where widget can be placed.

hmsjy2017, 2020-10-31:
For Windows and Linux, I think you can put a button in the taskbar and open the widget through this button. Not long ago, [Swift was available on Windows](https://www.infoworld.com/article/3583588/apples-swift-language-comes-to-windows.html)

exelban, 2020-10-31:
I know that swift can be compiled under Windows. But my app is a UI app in 75%. And there are no good UI frameworks available on Windows. But this app is about monitoring in the menu bar. So I think it will be available only on macOS.

closed 2020-11-01 by exelban

hmsjy2017, 2021-04-08:
I think it might be possible to use OpenSwiftUI instead of SwiftUI on Windows and Linux platforms.
https://github.com/Cosmo/OpenSwiftUI

doreks, 2022-04-13:
> I think it might be possible to use OpenSwiftUI instead of SwiftUI on Windows and Linux platforms. https://github.com/Cosmo/OpenSwiftUI

would that mean that this app could be used on linux?
How should we migrate it or make it functional?

rajatgupta1998, 2024-05-28:
Thanks for the continuous development and support on this software. I use this actively on my M2 Pro MBP. 

However, as per above discussion, the app seems to support mac only. Is there a recommended alternative to be used on Linux (Ubuntu) ? 

Apologies for re-opening this! @exelban , do we have a community or forum for this app?

exelban, 2024-05-28:
@rajatgupta1998 I'm using htop on linux)

Nope, there is no something like that. I was trying to open a discussion. But for some reason, it does not go well.