exelban/stats #131: Question: runAtLogin

shoopdawoop, 2020-10-30:
Is there a way to initialise runAtLogin via command line (and what exactly does setting `runAtLoginInitialized` do)?

It would make automatic deployment a bit simpler.

exelban, 2020-10-30:
Hi.

`runAtLoginInitialized` variable allows detecting if the app was launched for the first time. It needs to turn on a launch at login as a default value.

exelban, 2020-10-30:
I don't really understand what you propose. This flag allows to launch the app when login to the system.

shoopdawoop, 2020-11-03:
Thank you. To clarify, when I deploy the app via an automated process, say with an installer script when setting up a new machine for a user, I would like to run a command in an install-script that activates the auto-login (and additional settings) and then start the app so the user does not have to do anything.

However, if I create the preferences file before the App is launched for the first time and set `runAtLoginInitialized` via `defaults write` commands it seems to be ignored by the app (neither `TRUE` nor `FALSE` seem to do the trick, as reflected by the GUI) while e.g. setting `update-interval` to `Never` seems to work.

I was wondering if maybe there was another way to do this, perhaps a command line option of the Apps binary that would let me set this?

exelban, 2020-11-03:
@shoopdawoop do you want to use it in your own app?
Or you want to change something in this one?

exelban, 2020-11-03:
This flag is not related to the installation process. And it's not a part of the flow when the app runs.

exelban, 2020-11-03:
When the app starts for the first time it checks if `runAtLoginInitialized` is true. If yes, it will start a `launchAtLogin` process. Which will be available only after the start of the app. 
It's valid only for the first start. This key can be false only if you never have started the app. Never else.

If you want to install this app to some mac without running. And have it opened after the next login. You need to run the `launchAtLogin` part manually (it can be found in the Stats.app). But some parameters (`SMLoginItemSetEnabled`) in macOS must be set to true.

I suppose the best way is to create a small script that will do it automatically after installation. But it must be run manually.
Maybe there is another option. I will need to google it)

shoopdawoop, 2020-11-05:
Thank you, with that bit of information I was able to solve my problem.
I did a little more testing and now got it to work the way I wanted.

I am basically creating the plist like this before the first launch:

```
{
  "update-interval" => "Never"
  "version" => "2.4.0"
}
```

What did the trick was indeed to not set `runAtLoginInitialized` at all (and which I tried earlier).

I also found that `version` has to be set or else the plist is being reset.
An older `version` works fine, as it gets updated with the current `version`, so I will keep doing that.

Not sure if I would call this as a bug, but should the `version` not just be added to the prefs at first run if it is missing?
This was one reasons I got confused earlier while trying to make this work.

Also, in all of my tests, the GUI never showed the "Login at Start" as "ON" during first launch, only after a restart of the app.

Again, thank you!

closed 2020-11-05 by shoopdawoop

exelban, 2020-11-05:
@shoopdawoop I'm just so interested. Can I ask what do you do?)

shoopdawoop, 2020-11-07:
The reason I am doing this is for automated deployment during new Mac Client setups where I might put Stats on as a default or optional install. This could be just a bunch of scripts running during a semi-automated process, e.g. from a mounted external drive or server volume, or with a managed service such as [Munki](https://github.com/munki/munki).

Instead of wasting a minute each time to manually install and setup the app, I can now run a simple script that does the job (mount the dmg, copy the app, eject the dmg, setup the prefs and launch the app). It makes little sense for the average user, but makes an admins life easier.

exelban, 2020-11-07:
@shoopdawoop interesting project. Thanks for explaining)