exelban/stats #134: recycle memory

sumanthratna, 2020-10-31:
(feature request)

Right now I use [Memory Diag](https://apps.apple.com/us/app/memory-diag/id748212890), [Fanny](https://github.com/DanielStormApps/Fanny), and [Battery Monitor](https://apps.apple.com/us/app/battery-monitor-health-info/id836505650). I'd like to use stats but I'm unsure if it has a "Recycle memory" feature (Memory Diag has this).

When clicked, Stats should clear up some memory. I'm not sure how this works, but I'm fairly confident it isn't just a `sudo purge` because I'm never prompted for my password.

exelban, 2020-10-31:
Hi.

This app is using only for monitoring. But I think this feature could be added. 
But I don't know what exactly `Memory Diag` do. Does it just clean the RAM cache? Or something else?

sumanthratna, 2020-10-31:
hi! here are two screenshots of Memory Diag:
| [image] | [image] |
| --- | --- |

as far as I know, it only shows a pie chart and has the "recycle memory." In order to implement this, I think we could just use up a bunch of memory and then clear it? e.g. https://stackoverflow.com/a/6840277/7127932. To do this in Swift, this may help: https://developer.apple.com/documentation/coremedia/cmmemorypool-u89

It might be possible to reverse engineer Memory Diag but I probably can't help with that

exelban, 2020-11-01:
Ok. I will take a look at this.

Military-Doctor, 2020-12-07:
In my opinion, recycling memory is not necessary. 

When you recycle memory, the operating system collects it and stores it on the hard disk (HDD or SSD). Once you want to reuse those memory, the operating system retrieves the data from the hard disk. (See virtual memory implementation)

It will cause redundant I/O, lots of data travels on the road, from hard disk to memory to CPU. We call this road the `bus`.
Hard drives are slower than memory, therefore, the performance of the program will have significantly impact. 

I think it's better to use much more of the memory instead of freeing it up.

exelban, 2020-12-07:
@Military-Doctor I agree with you. Recycling memory is not necessary. At least in macOS.
macOS have great memory management (for me). And I prefer to keep this management on macOS.

But I think there is a reason to have this button. I don't have a problem with RAM now. But when I have 8GB mac. It was a problem. From time to time macOS was not fast enough to clean up the RAM of the closed app. And a button like this will solve this.

Military-Doctor, 2020-12-07:
@exelban This is a good scenario to explain why we need this button.

exelban, 2021-01-17:
Hi. I need some help with this.
I really want to add this feature. I've already spent some time trying to implement this. But option with [allocating memory](https://stackoverflow.com/questions/6755954/how-do-os-x-applications-clean-memory/6840277#6840277) till macOS starts releasing RAM does not work well. And also takes too much time.

Maybe someone has an idea of how it could be implemented?

sumanthratna, 2021-01-18:
[From the Memory Diag FAQ](https://rockysandstudio.com/support.html#faq-md-optimize:~:text=The%20recycle%20of%20Memory%20Diag%20tries,app%20and%20in%20several%20other%20scenarios):
> The recycle of Memory Diag tries to restore the system’s memory state to the state of a “fresh system restart”, mostly by forcing the system to empty caches and lower compressed memory. This only speeds up your system in special scenarios:
> 
> 1. After closing a memory intense app, it will clean up the caches left behind. In case if you know you will not start a memory intense app again in the future an optimization helps your system.
> 2. Before planing to run memory intense apps.

and [this Reddit comment](https://www.reddit.com/r/mac/comments/2hefvo/memory_clean_vs_memory_diag/ckrwuj9/?utm_source=reddit&utm_medium=web2x&context=3) says:
> Both utilities work by requesting enormous allocations from the system, forcing it to drop any optimizations it may have made in terms of objects stored in RAM as cache, or worse, forcing it to swap other allocations to the hard drive.

@exelban perhaps try changing the amount of memory you request? My guess is that iterating over this process might work a little better than requesting a big chunk of RAM and then clearing it:
1. request a small amount of RAM (maybe 20 to 40 MB?)
2. clear the requested RAM
3. if the resulting RAM level decreased by less than 5 MB, break because trying again probably won't help

if that doesn't work, you could try running `sudo purge` and asking the user for their password

Military-Doctor, 2021-01-18:
@sumanthratna , Try `sync && sudo purge` manually

exelban, 2021-01-20:
Hi again.
I finally figured out what I can do with this.

**TL.DR:** Cached Files is not used RAM.

First of all. All apps (like Memory Diag) just free cached files from RAM (not from VM as @Military-Doctor writes, cached files have nothing to VM). Cached files have one purpose: to speed up your mac. Not slow his down. And by cleaning this cache, some of the apps could run slowly (for sure if they were closed before). 

Example.
Your mac has 4GB of RAM. 2GB takes macOS. And you have 2 free. You running 2 apps: `A` and `B`. `A` takes 1GB. `B` takes 1GB. You have no free RAM. Now you closing app `B`. And in theory, you will have 1GB free after closing. But not exactly. Because macOS move `B` from the application part to cached. So. What if you will want to run some app. If you run again app `B`. It will run super fast. Because it will not require a new allocation. Will be enough to move it from cached to application. And if you clean the cache, you will not receive this speed up.
So what if you want to run the app `C` which needs 500MB. macOS just rewrite the 500MB in the cached part of RAM. In this case, you could run the app `C`. But even if you will close it and run `B` again. Running the `B` could be a little bit faster. Because some part of memory is already in RAM (only 500MB from 1GB was purged by `C`).

The main conclusion for me: Cached Files is not used RAM. And I don't really understand why some apps (Memory Diag again) show it's as used. And purging this RAM could only slow down my mac in most cases.

If for someone it will be more helpful: it more like how SSD (correct me if it's a way how HHD, but I think it's an SSD because of limited IO) works. SSD doesn't delete the file when you deleting it from OS. It will just be marked as `ready for reusing`. And when new data will come, they will just replace the deleted file.

For people who think that `purge` really speed up their mac. I would highly recommend using `Memory Pressure`, not `Memory Usage`. To check if you have enough RAM or if macOS has some problems because of RAM. Because in these case usage it's not the best indicator. It's normal to have even >90% of RAM usage on macOS. It does not mean that your mac cannot run an additional app or some app freezing because of this.

I suppose that a lot of people (also I) take a look at memory usage, saw that it >50% + they mac does not have any app running. And they think that there is some problem. Because OS takes too much RAM. But macOS (as UNIX) using RAM in a little bit different way than Windows.

End of story: I don't see any reason to add the `Recycle` button. Because it has no sense and just confusing the users. Giving them an idea that this button will clear their RAM.

closed 2021-01-20 by exelban