exelban/stats #135: Fan Monitoring Not Working

MyAudioDNA, 2020-11-01:
When I turn fan monitoring on, I just get a white line in my status bar and as soon as I click away, even the white line disappears. It looks exactly how the Sensors monitor looks if you turn it on without actually turning on any sensors.

[image]

I'm on a 2019 16" Macbook Pro and my OS and software are all up to date.

exelban, 2020-11-01:
Thanks for reporting this.

Could I send you a debug version of the app tomorrow to test if fans will be available?
It looks like in new macbooks fans have different smc key, or missing some parameters.

MyAudioDNA, 2020-11-01:
> Thanks for reporting this.
> 
> Could I send you a debug version of the app tomorrow to test if fans will be available?
> It looks like in new macbooks fans have different smc key, or missing some parameters.

Absolutely! I'd be happy to test it for you.

jgoldhammer, 2020-11-02:
I have also activated the fans section and it worked for me. BUT: After enabling the fans section stats crashed after some seconds each time I have started it. After deactivating the fans information stats now is working again.  It is possible to deliver log information?

exelban, 2020-11-02:
Hi.

Could you try this one?
[Stats.dmg.zip](https://github.com/exelban/stats/files/5473688/Stats.dmg.zip)

iliakukharev, 2020-11-02:
Sensors and Fans are working on this build
Sensors is not that obvious, as you need to select the sensor after enabling the general switch

exelban, 2020-11-02:
I'm not so sure. Some macs can have different keys or some of them can miss some fan params (max/min speed etc).
All sensors are disabled by default. But in Fans module all fans are enabled by default.

MyAudioDNA, 2020-11-02:
> Hi.
> 
> Could you try this one?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/5473688/Stats.dmg.zip)

In this build, I do see the fan data showing now. However, I'm not sure it's accurate. It is currently showing both fans spinning at approximately 3000 RPM, which is about 55% capacity, but there is basically no fan noise whatsoever. If both fans were running that high right now, I would think there would almost certainly be at least some fan noise.

iliakukharev, 2020-11-02:
3000rpm on 16" Pro is almost silent, btw
in my case numbers are the same as mac fans control shows

exelban, 2020-11-02:
Please do not update to the latest version. It has disabled the Fans module (#136).

MyAudioDNA, 2020-11-02:
> Please do not update to the latest version. It has disabled the Fans module (#136).

Understood. Thanks!

exelban, 2020-11-08:
Fans module available again in the [last](https://github.com/exelban/stats/releases/tag/v2.4.2) release.

closed 2020-11-08 by exelban