exelban/stats #136: Stats 2.4.0 crashes after each app restart

jgoldhammer, 2020-11-02:
Stats 2.4.0 crashes for me. After a restart, stats crashes every time.
I have attached a crash report...

[Stats_2020-11-02-084010.txt](https://github.com/exelban/stats/files/5473391/Stats_2020-11-02-084010.txt)

exelban, 2020-11-02:
Hi.

Thanks for reporting. Could you try this version?
Could you also provide your mac specs? Because it looks like some memory bug.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5473579/Stats.dmg.zip)

Coliinnn, 2020-11-02:
I can confirm this has happened to me as well since updating to 2.4.0 yesterday. [These are the specs of my mac](https://i.imgur.com/ykSPwPK.png). The crash still happens with the DMG you just provided on my Mac.
[Stats_2020-11-02-095250.txt](https://github.com/exelban/stats/files/5473738/Stats_2020-11-02-095250.txt)

exelban, 2020-11-02:
Thanks.
Could you try this one?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5473772/Stats.dmg.zip)

Coliinnn, 2020-11-02:
So far this seems to work. I have it running about ten minutes without a crash, which seems promising

exelban, 2020-11-02:
It crushes only on start. So, after the app will work normally.

Unfortunately, I don't fix a problem with this build. I only remove the problem place. 
I cannot reproduce the problem on my computers. Could maybe someone check a few more builds? To ensure that problem is fixed.

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5473907/Stats.dmg.zip)

Coliinnn, 2020-11-02:
This build looks good too on my end

exelban, 2020-11-02:
@Coliinnn thanks. Could you check the sensors and fans?
If the widgets show correctly. Because the problem was in these widgets.

Coliinnn, 2020-11-02:
I just came back to my computer and it did crash in this latest build. I was already using the fans widget, not the sensors one. The fans widget was showing correctly at least, but it did result in a crash seemingly. Is there a specific sensor to test or can I try any?
[Stats_2020-11-02-105053.txt](https://github.com/exelban/stats/files/5474052/Stats_2020-11-02-105053.txt)

exelban, 2020-11-02:
Nope. If it crash, it will crash. The widget is the same in sensors and fans modules. Application crashes when trying to calculate what width of widget must be.

exelban, 2020-11-02:
Build nr.1: [Stats.dmg.zip](https://github.com/exelban/stats/files/5474107/Stats.dmg.zip)
Build nr.2: [Stats.dmg.zip](https://github.com/exelban/stats/files/5474120/Stats.dmg.zip)

exelban, 2020-11-02:
Don't post a crash report. It has the same bug and the same info.

Coliinnn, 2020-11-02:
Ok, I didn't know. So the issue is not even part of the fans or sensors themselves. Build nr.1 crashed for me, as did build nr.2

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5474340/Stats.dmg.zip)

wfrank21, 2020-11-02:
Crashes for me as well. (all builds mentioned here)

Coliinnn, 2020-11-02:
Yep, even this last one from four minutes ago

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5474400/Stats.dmg.zip)

wfrank21, 2020-11-02:
Crashes

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5474456/Stats.dmg.zip)

Coliinnn, 2020-11-02:
This too crashes

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5474604/Stats.dmg.zip)

exelban, 2020-11-02:
Sorry for the long responses. But I must wait till Apple validates and sign a new build.

Coliinnn, 2020-11-02:
That's ok. I (and others too I'm sure) already appreciate your responsiveness and attempts at fixing this. I see the last build has changed the fan module display; from two to one row. So far no crash yet and I restarted it three times to test.

wfrank21, 2020-11-02:
Works for me too! Donation will be made! Thanks alot!

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5474961/Stats.dmg.zip)

wfrank21, 2020-11-02:
In this one only the left side shows up and crashes after around 30sec-1min.

exelban, 2020-11-02:
Hmm, could you send a crash report?

wfrank21, 2020-11-02:
[Stats_2020-11-02-133247.txt](https://github.com/exelban/stats/files/5475010/Stats_2020-11-02-133247.txt)

exelban, 2020-11-02:
Does the last build works?

wfrank21, 2020-11-02:
The one from 2 hours ago works. This one only shows one fan (left) but in normal size.

exelban, 2020-11-02:
Oh, I know that some version works. Could you try the last one and say if it works or crashes?

wfrank21, 2020-11-02:
Last one (13:29, CET) crashes. Log from 13:38 CET relates to the last one.

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5475305/Stats.dmg.zip)

wfrank21, 2020-11-02:
14:26 CET crashes as well.

exelban, 2020-11-02:
Thanks for helping. I have a few macs where I can test an app. But on all it works fine. So I'm trying to fix the problem. But I cannot reproduce it(
So the only option is someone who has crashing the app, will check if it still crashes.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5475428/Stats.dmg.zip)

wfrank21, 2020-11-02:
14:49 CET seems to work fine. But now no Sensors or Fanspeeds show up.
[image]
[image]

exelban, 2020-11-02:
Yeap, I know exactly where is the problem. I just don't know how to fix it. App crash because it tries to access the variable which is free. But it doesn't use pointers in this part of the app. And it makes a problem to fix it.

exelban, 2020-11-02:
This build will show the sensors if you switch `Display mode` to `One row`.
[Stats.dmg.zip](https://github.com/exelban/stats/files/5475505/Stats.dmg.zip)

Coliinnn, 2020-11-02:
That's weird. The build from a few minutes ago runs fine for me, no crash. One row fans displays them next to each other

exelban, 2020-11-02:
Yeap, it's if you have switched to `One row`. The problem appears only when it tries to draw 2 rows.
[Stats.dmg.zip](https://github.com/exelban/stats/files/5475561/Stats.dmg.zip)

exelban, 2020-11-02:
Build nr.2: [Stats.dmg.zip](https://github.com/exelban/stats/files/5475590/Stats.dmg.zip)

Tell please if some of these not crash.

wfrank21, 2020-11-02:
Build from 15:14 CET does not crash, but only shows fan speeds in one row mode.
Build from 15:20 CET crashes with two rows.

Coliinnn, 2020-11-02:
Nr1 didn't crash but also did not draw two rows. Nr2 crashed with two rows. I see wfrank has the same just now

exelban, 2020-11-02:
Thanks. What about this one?
[Stats.dmg.zip](https://github.com/exelban/stats/files/5475737/Stats.dmg.zip)

Coliinnn, 2020-11-02:
This crashed for me

exelban, 2020-11-02:
Fck, I can't understand why it releasing the objects.
Could you maybe disable the Fans module and check if change something?

wfrank21, 2020-11-02:
Crashed for me as well. Trying now without Fans Module

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5475895/Stats.dmg.zip)

wfrank21, 2020-11-02:
15:41 CET works without Fans Module.
Trying 16:04 CET now

exelban, 2020-11-02:
Really? The app which crashes works if the Fans module is disabled?
It can change a lot)

wfrank21, 2020-11-02:
Correct.

wfrank21, 2020-11-02:
16:04 CET seems to work fine, but only one row mode works

exelban, 2020-11-02:
Could you try this one with the disabled Fans module and enabled Sensors?
[Stats.dmg.zip](https://github.com/exelban/stats/files/5476016/Stats.dmg.zip)

Coliinnn, 2020-11-02:
16:04 when the fans module is enabled it shows just an empty space in the menubar for two row. 
[image]. No crash though. Single row displays it the right way

exelban, 2020-11-02:
[16:04] This version works because it has disabled rendering the 2 rows.

entee55, 2020-11-02:
16:04 CET works for me but no sensor readings. It only shows blank

exelban, 2020-11-02:
Yeap, it works and shows blank because it has disabled rendering (which is the crashing  reason).

exelban, 2020-11-02:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5476077/Stats.dmg.zip)

Coliinnn, 2020-11-02:
16:17 with fans disabled and two-row sensor heatpipe readings enabled does not crash for me it seems

entee55, 2020-11-02:
16:17 Sensors work with 2 row, and fan is also seems to be good, but I only have one fan. The 2 row view makes my 1 fan go on top. No crashes yet.

[208 more replies]

closed 2020-11-08 by exelban