exelban/stats #138: Feature request: customise 'n/a' battery time remaining

tech189, 2020-11-03:
I think it would be quite nice if we could change the text that the battery menubar icon displays when it is either calculating the remaining charge time or when it is full. In my opinion 'n/a' is vague and not very descriptive, I would prefer something like (...) or (···) or maybe even an hourglass wait sign (⌛︎) (⧗) for when it is calculating and (full) for when it is fully charged but still plugged in to AC.
Screenshots of when calculating and when full for reference:
[image]
[image]

Thanks for this great app otherwise!

exelban, 2020-11-03:
Hi. Thanks. 

I don't like this `n/a` too. I can change it to something. I don't know if these symbols will be good for users. Maybe 3 dots will be better.

I will run the app with all these options and post screenshots of them. It will help to decide.

exelban, 2020-11-09:
[image]
[image]
[image]
[image]

tech189, 2020-11-09:
I think the three dots look the most elegant for when it is calculating the time remaining! Can we have 'full' for when it is 100%? Or maybe just hide the additional time at 100%?

exelban, 2020-11-09:
Yeap, for me dots look better than symbols. What do you mean `full`? Like text?
I think about hiding the additional time when it's 100%. Previously additional information hides if the battery was full. But it looks like this behavior was deleted after rewriting (v1->v2).

I propose to leave `n/a`. And add a settings option `Hide additional information if full`. This option will hide all additional information if the battery is fully charged. What do you think?

exelban, 2020-11-09:
Just because dots look like the text is not sized in the frame. It just looks like something is broken or some bug.
My feeling after all day with 3 dots)

tech189, 2020-11-10:
I think that an option for hiding additional information if full is a good idea. I originally thought that the 3 dots could be displayed for the few seconds while the app calculates the time remaining but I agree that 3 dots being constantly displayed (such as when full) is not good because it does look like something is missing.

In summary, I think it should only show the three dots when calculating, and when full it should disappear entirely.

Not sure what would be good if you have both percentage and time as additional information. Either both disappear or only the time remaining disappears when the battery is full.

That way showing `n/a` is avoided entirely.

exelban, 2020-11-15:
Hi.
I [implement](https://github.com/exelban/stats/releases/tag/v2.4.3) the option to hide the additional information when the battery is full.

I leave the `n/a` for now. I'm just not sure that 3 dots is a better option.

closed 2020-11-15 by exelban

s-leeq, 2024-07-29:
Sorry for re-open this thread, but maybe "Calculating" and "Not Charging" texts during these two status might be a little bit better? I am using AlDente and my battery is never been full, so I can't use the hide function while my laptop is not charging. Another app "iStat Menus" is able to handle this well, here are the screenshots:

[image]
[image]