exelban/stats #139: Feature request - Please make an option to show power (wattage) consumption Integer

hoangtocdo90, 2020-11-03:
Thanks for making great software.
Please make an option to show power (wattage) consumption Integer. Power consumption in the menubar in decimal is a waste of space.  In my case Power consumption fluctuate from X.XW to XXX.XW and it makes the menu bar shrinking some space, and another menu icon moving. Or W symbol turn to V 
[image]

exelban, 2020-11-03:
Thanks.

I will fix it.

exelban, 2020-11-08:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.2) release.

hoangtocdo90, 2020-11-08:
I think it's not done yet. Please take a look at this video, Power consumption fluctuates from X.XW to XXX.XW and it makes the menu bar shrinking, and another icon move.
I think you should calculate space for a maximum of 999W and remove the final decimal.
https://1drv.ms/v/s!AmzaUwpwHutPgqMfR5cQNV3UCEWcAA?e=ctdso5

exelban, 2020-11-08:
Yeap, my fault. 
There were problems with sensors modules. And at some moment I revert the changes. Will be fixed in the next release.

exelban, 2020-11-15:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.3) version.

closed 2020-11-15 by exelban