exelban/stats #140: GPU usage not shown correctly in menu bar

entee55, 2020-11-04:
**Describe the bug**
GPU usage shows 0% in menu bar, sometimes it jumps to about 10%. It should be around 100% since I am rendering 4k video.  Popup shows 2 rows (don't understand why) one is at full speed. In menu only one GPU option is visible.
I do not know how it should work, maybe my gpu cores (if it has more) identified separately.

Maybe I am wrong and it supposed to be like this, then please explain.

[image]

**Details:**
 - Device: MacbookPro 13" Early 2015
                 2,7 GHz Dual-core Intel Core i5 
                 8 GB 1867 MHz DDR3
                 Intel Iris Graphics 6100 1536 MB
 - MacOS 10.15.17
 - Application version: Last version in No fan issue debug feed.

exelban, 2020-11-04:
Hi.

Thanks for reporting. GPU reading it's a little bit strange process. There is only 1 option to read GPU stats (I don't find any other). The drivers report a value and I can fetch these values. If something shows wrong it means that the driver reports strange values. 

It's not the first time, you can check #121. There is some description of how it works, and what can cause the problem.

TR.DR. I don't have any influence on these reads. It means that the system reports these values and I cannot fix it.

exelban, 2020-11-04:
You can help to debug why it has a strange usage.
Post please of `ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt`. 
This command makes the same read that the app does. Open please some youtube video, or something that must load the GPU and the app shows a wrong value. It helps to say where the problem could be. In the app or in the driver.

exelban, 2020-11-15:
Closed due to inactivity.
Please reopen this issue, If someone would like to help with this.

closed 2020-11-15 by exelban