exelban/stats #141: Fans disappear and come back in menu bar

entee55, 2020-11-04:
**Describe the bug**
 I noticed that the fan RPM disappeared again like I've said it before. I can click next to the other module and pops up RMP: 1478.
[image]

This screenshot was taken when holding click on the thin line.

After abut half hour I noticed to change back:
[image]

Sometimes it goes to white background and after a while back to normal:
[image]

**Details:**
Device: MacbookPro 13" Early 2015
2,7 GHz Dual-core Intel Core i5
8 GB 1867 MHz DDR3
Intel Iris Graphics 6100 1536 MB
MacOS 10.15.17
Application version: Last version in No fan issue debug feed.

exelban, 2020-11-04:
Thanks for reporting. I will take a look later today.

exelban, 2020-11-08:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.2) release.

closed 2020-11-08 by exelban