exelban/stats #143: eGPU doesn't appear at "GPU to show" dropdown

cadic, 2020-11-04:
**Describe the bug**
When I connect RX 460 external GPU, it does not appear in the list of GPUs. Replacing the card with RX 590 resolves the problem, but why RX 460 does not work?

[image]

[image]

**Details:**
- Device: Macbook Pro 15 2018
- MacOS 10.15.7
- Application version 2.4.1
- RX 460 through Mantiz Saturn Pro enclosure

exelban, 2020-11-05:
Hi. I'm not sure if the eGPU can be read in the same way as to build-in.
Post please the result of `ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt`. When connected `RX 460`. It will help to say why gpu is not visible.

cadic, 2020-11-05:
@exelban thanks!

[Stats log.txt](https://github.com/exelban/stats/files/5495385/log.txt)

[ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt](https://github.com/exelban/stats/files/5495391/output.txt)

exelban, 2020-11-05:
@cadic please try this one and give feedback if it shows all cards.

[Stats.dmg.zip](https://github.com/exelban/stats/files/5497226/Stats.dmg.zip)

cadic, 2020-11-05:
@exelban thanks, it doesn't fix the issue, I still just see builtin Intel and Radeon, no external GPU.

I will be able to swap the external card from RX 460 to RX 590 (it was displayed in "GPU to show") on weekend. Will provide `ioreg` report then, hope it will help to see the difference between GPUs

exelban, 2020-11-05:
hmm, it's strange. You must see it as `AMD Graphics` or `Unknown`.
I add debug mode to the app. Please run the app with an argument `--debug`. It will write a log to the file (as in the previous build).

To run the app with argument you need to run it from `Console` using this command:
`/Applications/Stats.app/Contents/MacOS/Stats --debug`.

It will say me why RX 460 still not visible.

exelban, 2020-11-05:
Yeap, but replace please `/Applications/Stats.app` with the last build path.

cadic, 2020-11-05:
@exelban latest 
[log.txt](https://github.com/exelban/stats/files/5497488/log.txt)

exelban, 2020-11-05:
Oh, I see. It detect both amd cards as the same model name.

```
Processing AMDRadeonX4000_AMDBaffinGraphicsAccelerator acceletator
Radeon Pro 555X: utilization=0, temperature=57

Processing AMDRadeonX4000_AMDBaffinGraphicsAccelerator acceletator
Radeon Pro 555X: utilization=0, temperature=63
```

And the second one just rewriting the first one. Give me a few minutes, I will think how it can be fixed)

cadic, 2020-11-05:
@exelban yeah, seems the card name is based on it's architecture, not model.

Just plugged **RX 590** and it works. The log displays different name for it: `AMDRadeonX4000_AMDEllesmereGraphicsAccelerator`

[Stats RX590 log.txt](https://github.com/exelban/stats/files/5497566/log.txt)
[ioreg output-rx590.txt](https://github.com/exelban/stats/files/5497572/output-rx590.txt)

exelban, 2020-11-05:
A quick test. Does it show RX 460 properly?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5497666/Stats.dmg.zip)

cadic, 2020-11-05:
@exelban yes!
(Notice that 555X turned into **AMD Graphics** now)
[image]

exelban, 2020-11-05:
Nice, nice.
Thanks for your help. 

Unfortunately, I cannot properly detect the model of GPU. But it still better to see AMD graphics than nothing.
This fix will be available in the next release. But you can use this build if you want.

cadic, 2020-11-05:
@exelban thank you!

exelban, 2020-11-08:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.2) release.

closed 2020-11-08 by exelban