exelban/stats #145: Fanspeed Widget doesn't show up

wfrank21, 2020-11-04:
Hi!

As mentioned in #136: I have the issue that Fanspeeds are not showing up. It's mostly resolved by restarting the System.

[image]

I have a MacBook Pro 2017 with the 3,3GHz i5 and 16GB of RAM

exelban, 2020-11-04:
Thanks.
Do you see any sensors in the settings?

wfrank21, 2020-11-05:
[image]
This is the fan settings page.

exelban, 2020-11-05:
Thanks. So it looks like it's the same issue as #141 (you also use the same app build). 
I will close this issue.

closed 2020-11-05 by exelban