exelban/stats #150: feature request: sort network processes by combined up+down

stewartadam, 2020-11-10:
**Describe the bug**
Process sorting appears to sort by download rate, which means if an app is uploading a lot, it jump around the list (or might not show up at all). I recommend combining the process' upload/download rates and sorting by that :)

**Details:**
 - Device: MBP 2016 15"
 - MacOS 10.15.7
 - Application version 2.4.2

exelban, 2020-11-10:
Hi.
The network top processes are sorting by upload and download values.

```swift
            if $0.download != $1.download {
                return $0.download < $1.download
            } else if $0.upload < $1.upload {
                return $0.upload < $1.upload
            } else {
                return $0.time < $1.time
            }
```

It means next:
- sort by download values if not zero
- else sort by upload values if not zero
- else sort by time

stewartadam, 2020-11-11:
The challenge here is high TCP upload traffic implies some (small) level of download due to TCP traffic, so by having a non-zero download an uploading process doesn't necessarily get sorted near the top of the list.

exelban, 2020-11-15:
I don't understand. Maybe example will help me to understand the problem:
- app1: d=100 u=0
- app2: d=50 u=50
- app3: d=0 u=100
- app4: d=10 u=10
- app5: d=10 u=20

The order of processes will be next: `app1, app2, app3, app5, app4`. What's wrong with this sorting?
If I implement some rate (upload/download). Then the order will be next: `app3, app2, app4, app5, app1`. I suppose some more complicated rate must be added to sort this. But I don't see a real reason for this.

The list is showing the processes with the most usage of networks. Doesn't matter if it's a high upload or download traffic. And yes, if some process has d=10 u=0 and the second one d=3 u=20. The second one must be on the top of the list.

stewartadam, 2020-11-16:
> The order of processes will be next: app1, app2, app3, app5, app4.

What I see is `app1, app2, app4, app5, app3`. I believe your comparator snippet above will never compare two upload speeds unless their download is equal (app4/5), so app3 is always last.

exelban, 2020-11-16:
Hmm, yes. You right. I will take a look at this sorting and try to fix it. Thanks.

exelban, 2020-11-17:
Hi.

I update the sorting algorithm. Now it finding the max and min value between 2 processes. And now:
- if min == min -> sort by last usage time (it's possible only for processes with 0 download and upload)
- if max == max but min != min -> sort by minimum value because the max are the same
- else sort by max value

It allows sorting processes by the maximum value (download or upload). And if there will be 2 processes with the same max value (let's say it will be download for example). But the minimum value (upload in this case) is different. The processes will be sorted by minimum value.
If both values are the same (upload and download). It will sort by the last usage time.

So my example processes:
- app1: d=100 u=0
- app2: d=50 u=50
- app3: d=0 u=100
- app4: d=10 u=10
- app5: d=10 u=20

Will be sorted: app1, app3, app2, app5, app4. Does it make sense?

stewartadam, 2020-11-17:
Sounds great!

closed 2020-11-22 by exelban