exelban/stats #152: Feature Request: Fan control

cgarrovillo, 2020-11-11:
I think I've seen this here, not sure where it went. Now that fan monitoring is up,  controlling fan speeds should be the next step.

exelban, 2020-11-11:
Hi.
There was an issue. But it was closed due to fan visualization was added. 

I'm not sure if I want to add some manipulation to the app. I hope it will be just a monitoring app.
But, I think it could be add in the future. But it will be not the main priority for sure.

exelban, 2020-11-14:
If someone wants to make it possible. You can help with UI/UX. 
I don't know how it can be implemented and where it can be placed. In settings? Or in the popup?

kevinm6, 2020-12-28:
I think the best way could be in the popup, on the top of the list of the “Sensors Menu”, even with a simple slider and the value.
Maybe even adds some fan speed preset based on a sensor temperature, like:
    CPU Proximity ≥ 60 °C  ——> Fan 2000rpm (25%)

exelban, 2021-01-10:
@kevinm6 thanks for propose. 
It can't be just a slider. There must be a few elements: reset to default button, max button, min button, and slider.
Also, it will be not in the Sensors module for sure. It will be in the Fans module.

kevinm6, 2021-01-10:
@exelban I personally don’t use the “Fans” module, because for me is kinda of duplicate of the “Sensors” module, and I have the fans that shows on the Sensors module.

You could add some preset for fan speed and yes, a reset to default button, but event with a simple segmented control.

exelban, 2021-03-09:
Hi everyone who interested in this feature. I finally find some time to implement it.
And the first problem appears. Changing the fan speed requires root. And the app must be restarted. I see 2 solutions:
- not implementing this feature, because it requires root. And I tried to leave the app working without root
- it will ask you for a password when you will try to change the fan speed for the first time. The app will restart with root permissions. After you will able to change the fan speed

Which option is better?

cgarrovillo, 2021-03-09:
> * it will ask you for a password when you will try to change the fan speed for the first time. The app will restart with root permissions. After you will able to change the fan speed

I think this would be the best way. That being said to keep it noob-friendly, the permission request should only come from trying to change the fan speeds for the first time (like you said).

closed 2021-03-28 by exelban