exelban/stats #159: Temperature shows CPU core 8 when there are only 6

finnrobinanders, 2020-11-13:
The temperature dropdown view shows the correct 6 CPU cores and then an extra one labeled CPU core 8 on my 2019 Macbook Pro with a 6-Core Intel Core i7. The CPU core 8 displays a constant 264˚ F. 
[image]

exelban, 2020-11-14:
Hi.
Thanks for reporting. Unfortunately is not related to the app. Normally macOS not provide a sensor if it does not exist. But in your case macOS return a max value. It happens from time to time in some macs.
I see only one solution, I can remove the sensor if the temperature more than 250F (~120C).

exelban, 2020-11-14:
It was already the same issue #148. And there is a logic that removes the temperature sensor from the list if the SMC returns more than 250F (120C). But it runs only on the initialization step. It means, that on the initialization step SMC reports a value less than 264F. 

This logic (checks if temperature more than the threshold and remove if true) could be run when updates the values. But I'm not sure if it's the right way. Because (in theory) some sensors could report so high temperature.

finnrobinanders, 2020-11-14:
Okay, thanks for the explanation. Macs Fan Control also shows all the temperatures and it does not show CPU Core 8.

Suggestion: I think a better solution would be to check whether the mac is a 6 core or 8 core Macbook Pro on startup and only display the corresponding number of CPU temperatures.

exelban, 2020-11-14:
I'm not sure if it's a good solution. 

App automatically detects all sensors in the system. And normally the sensor must be invisible if the temperature sensor report more than 110C when the module initializing. After initialization, it's not possible to remove the sensor.

exelban, 2020-11-15:
Please reopen the issue, If you have a solution for it.

closed 2020-11-15 by exelban

Military-Doctor, 2020-12-07:
This is a bug and already fixed. #170