exelban/stats #160: Feature Request: Option to show disk space remaining instead of space used

wzxu, 2020-11-15:
In the popup menu of disk usage, I think "space remaining" is a more useful information than "space used".
Of course it's better to make it an option so different users can choose whichever they like. Thoughts?

[image]

exelban, 2020-11-16:
Hi.
I prefer to have all information. The used space, free space, and total space. Unfortunately, it will not fit in the view like this. So I decided to show a user space and a total space. In the same way, the spacing presented in the default macOS disks preview.

I'm not sure that adding an option to select the preferred spacing (free or used) will be a good solution.
I prefer to redraw the disk view and show all spacing.

For now, I can add a toolTip to the bar. So when your mouse will be over the disk bar. The toolTip with free space will show.

wzxu, 2020-11-16:
I got the idea of showing free space from [Monity](https://monityapp.com/), which I loved to use before upgrading to Big Sur. 
Just FYI, how Monity handles this is to toggle between free/used space when you click on the text.
Might not work well with Stats, though, since clicking already opens the Finder to the volume.

Anyway, a tooltip sounds pretty nice. Thanks for considering it!

exelban, 2020-11-16:
Hmm, thanks for the idea with a click. I think it can be implemented.
I can move the open disk from all disk view to only the top (name) and the middle (bar) part. And bottom part (disk spacing) will change free/used space.

So there will be 2 click actions:
- 2/3 of height will open the disk
- bottom 1/3 will change free/used space

What do you think?

wzxu, 2020-11-17:
It might work if there's a clearer separation between the clickable parts, I guess. I'm not that good in UI design 😛

P.S. I just found out Apple actually shows free space by default in the About Mac window, with tooltip for each section.
[image]
Then there's the Disk Utilities, which shows both free/used parts with legends.
[image]

Something for your reference.

closed 2020-12-13 by exelban