exelban/stats #161: Inconsistent font among components

idiotWu, 2020-11-16:
**Describe the bug**
Not sure if this is a bug (or a hidden feature 🤣), the font used in "Sensor" seems to be different from others:

[image]
(look at the `0` in "Sensor" and the `0` in "CPU")

**Details:**
 - Device: MacBook Pro 13" 2020
 - MacOS: 11.0.1
 - Application version: 2.4.3

exelban, 2020-11-16:
Hi.

Yeap, it's a feature) The font in the Sensor widget and in the other widget is different. 
All widgets using system font (SanFrancisco by default). But in the Sensors widget, it's a monospaced version of the system font.

idiotWu, 2020-11-16:
Thanks <3

IMO it would be better to use the same font in all widgets, is there any reason to use a different font in Sensors widget?

[image]

exelban, 2020-11-16:
Yeap. 
The sensor widget could contain a lot of values. These values can be different. For example 12.0W and 12.1W (and a lot more values). The width of these two rows will be different (because of the different width of 0 and 1). The sensor widget dynamically changes the width. So if the value changes from .1 to .0. The width will be changed. And you will see a small jump in the menu bar (because the width of the Sensor widget was changed).
To avoid resizing when it's not necessary (12.0 and 12.1) I must use a monospaced font. It means that the width of 12.0W and 12.1W will be the same.

This problem is not relevant to the other widget, because they are not so dynamic. And depends only on 1 value.

idiotWu, 2020-11-16:
> To avoid resizing when it's not necessary (12.0 and 12.1) I must use a monospaced font.

This kind of small shaking is also observed in other widgets when the value changes, although it doesn't affect the width of widgets:

[image]

So what about using the same monospaced font in all widgets which will also keep the UI consistent?

exelban, 2020-11-16:
In this case, the width of values is changing. But not the widget width. And there is no jumping in the menu bar.
Nope, I will not change the font to monospaced in all widgets. Why? Just because I like more how not monospaced looks like.

So, yes. There will be different fonts (technically the font is the same (SanFrancisco)). Monospaced in Sensors widget (Sensors and Fans modules) and not monospaced in others.

idiotWu, 2020-11-16:
> Just because I like more how not monospaced looks like.

Oh, thanks for your explanation 🤣. Great app, by the way 👍.

It would be grateful if the users could change UI fonts in the future :).

closed 2020-11-16 by idiotWu