exelban/stats #164: Huge spacing between icons on Big Sur

wernerfred, 2020-11-16:
Hi, currently using Stats version 2.4.3 on macOS Big Sur version 11.0.1. Sadly since the update to Big Sur the spacing between the icons is way bigger than before. Is this something you could "fix" or is this a general Big Sur thing?

[image]

exelban, 2020-11-16:
Hi. 
This is a new feature of macOS Big Sur. Each element in the menu bar has an additional margin. And it cannot be disabled.

wzxu, 2020-11-18:
Sorry for hijacking the thread for a bit. 

I have dragged the Stats icons to the leftmost positions, but every time I restart my Mac, they will be relocated in between the icons from other apps and the system icons. (I think you can also see it from the picture above. The order will always be reverted to other apps, Stats, and then system icons.)

@exelban do you have any idea if this is a bug of Stats or is it also Big Sur that's doing this?

exelban, 2020-11-19:
@wzxu there are a lot of options for why the order is not saving. Does it always repeat with the same configuration of order?
Or you always trying the different order? And after restart it resets.

Another option, it's an app order start after the restart. But it must be handled by macOS.

exelban, 2020-11-19:
@wernerfred I'm going to close this issue.
I will make a little bit smaller internal margins of the widget. But it will be not so notisible.

PS: I hope Apple will provide some API for this spacing. But it's almost unreal(

closed 2020-11-19 by exelban