exelban/stats #166: Extra space in the sensors temp

gmcinalli, 2020-11-16:
Maybe there's an extra space between the sensor value and the "**°**".

[image]

**Details:**
 - Device: Macbook Pro 16" 2019
 - MacOS: 11.0.1
 - Application version: 2.4.3

exelban, 2020-11-16:
Hi.
It's not normal. I will take a look at this.

exelban, 2020-11-20:
Hi. Could I ask if you also see this spacing in the popup view?
[image]

gmcinalli, 2020-11-23:
Yes.
[image]

closed 2020-12-06 by exelban