exelban/stats #168: Temperature doesn't update in one of the screens in multi-monitor configuration

richardtop, 2020-11-16:
**Describe the bug**
When working with multiple monitors (e.g. an extra monitor and the built-in MacBook Pro screen, the temperature updates only on one of them, the active one).

**Details:**
 - Device: [e.g. Macbook Pro 2019 16]
 - MacOS [11.0.1]
 - Application version [2.4.3]

Active screen:
[image]

Inactive screen:
[image]
The temperature stays always at 70 degrees while all of the other indicators are changing their values.

exelban, 2020-11-16:
Hi.

Hmm, I rewrite a Sensors widget (#125). And now macOS decide when the values in the widget need to be updated. I suppose only the sensor's values not updating? Or all widget not updating?

richardtop, 2020-11-16:
[image]
The widget seems to work just fine, it's the menu bar value that get's stuck

richardtop, 2020-11-16:
Widget works perfectly, but the value in the menu bar is stuck (i.e. 70 C in this case)

exelban, 2020-11-16:
Element in the menu bar I called a widget) Each module (CPU, RAM, Sensors, etc) has its own element in the menu bar (widget). And my question is if only the Sensors are not updating? Or all of the elements (of Stats app) in the menu bar?

richardtop, 2020-11-16:
So far it looks like just the sensors component (or temp sensor) is not being updated.

exelban, 2020-11-16:
Yeap, thanks. I will fix it.

richardtop, 2020-11-16:
Highly appreciate, you've built a great product!

closed 2020-11-22 by exelban