exelban/stats #169: RAM panel shows incorrect top processes

Yoshi2889, 2020-11-16:
**Describe the bug**
The RAM panel does not show the most up-to-date processes as shown in the screenshot here (Activity Monitor is sorted by Memory and Safari was not running at the time):

[image]

**Details:**
 - Device: 13" Macbook Pro 2017
 - MacOS: 11.0.1
 - Application version: 2.4.2

exelban, 2020-11-16:
Hi.

What do you mean by `duplicate top processes`?
Does it shows always the Safari on the top of the list? 

Could you please run this command: `/usr/bin/top -l 1 -o mem -n 8 -stats pid,command,mem`?
It will show the top processes.

Yoshi2889, 2020-11-16:
Thanks for your reply.

As you can see in the screenshot, WindowServer is listed multiple times despite only ever having one process running.

The output is as follows:
[image]

This is -again- with Safari completely quit (not closed) for at least 10 seconds. 

While writing this comment, it seems to have gotten even more confused:
[image]

There is only one Safari and WindowServer process:
```
▶ top -l 1 -o mem -stats pid,command,mem | grep -e Safari -e WindowServer
53211  Safari           608M 
125    WindowServer     559M 
53218  com.apple.Safari 15M  
556    com.apple.Safari 12M  
2555   com.apple.Safari 7096K
53230  com.apple.Safari 4996K
53240  com.apple.Safari 3828K
675    SafariBookmarksS 3620K
2600   SafariLaunchAgen 1720K
623    SafariCloudHisto 1608K
2615   com.apple.Safari 1444K
```

exelban, 2020-11-16:
It looks like everything is ok. The reader just takes the output of this command. And visualize it.

Could you run `top -o mem -stats pid,command,mem`. And check side by side if the top processes in the popup and in the top are the same? It can be some latency because they will have different read moments.

Yoshi2889, 2020-11-16:
The list is in sync, until I quit Safari or some other process in the top list, and thus some of the top processes vanish. After this the top processes list becomes out of sync or duplicated.

Maybe there is some list structure in the app that should be cleared before every update?

exelban, 2020-11-16:
So after you close safari, it disappears immediately from the `top`, but not from the popup?

exelban, 2020-11-16:
There are no caching or saved lists.

Yoshi2889, 2020-11-16:
> So after you close safari, it disappears immediately from the `top`, but not from the popup?

Yes, but it also seems to happen at random. Safari just also got duplicated while it was running the entire time. It's just easier to trigger by closing Safari.

exelban, 2020-11-16:
I never saw behavior like this. Maybe there are some relations between this bug at the way how macOS works with memory.
I can't do anything with this. The reader just reads the value and visualize it. 
It could be some race. But it's almost impossible. The interval between updates is too big.

exelban, 2020-11-16:
I will take a look at this. But I can't promise to fix it.

exelban, 2020-11-16:
What RAM size do you have?

Yoshi2889, 2020-11-16:
Thank you, let me know if I can provide more data or help out in any way.

> What RAM size do you have?

My MacBook has 8GB RAM.

exelban, 2020-11-16:
Could you maybe run this build? I add a process id to the processes (as toolTip). Just stay mouse on some process and it must show the process id. Could you maybe check if the duplicated processes have the same pid?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5549828/Stats.dmg.zip)

Yoshi2889, 2020-11-17:
Sorry for not replying sooner. I tried your build, thank you.

When duplicated, all processes show the same PID. I tried taking a screenshot but this did not really work out as expected.

exelban, 2020-11-17:
No problem)

Thanks for the feedback. So it means that there is some racing. I will try to catch and fix it.

exelban, 2020-11-18:
@Yoshi2889 can I ask you to check this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/5562405/Stats.dmg.zip)

Yoshi2889, 2020-11-18:
Sure thing. Sadly nothing changed:

[image]

exelban, 2020-11-18:
Hmm, how many processes do you have selected in the settings?
8?

exelban, 2020-11-18:
According to this screenshot. The problem is not in racing. But in the reader.

exelban, 2020-11-18:
I must see it in the previous screenshots( But there were 7 from 8. And it was not so noticeable.

exelban, 2020-11-18:
Just a short description of why there are a duplicated and unsorted processes (on the last screenshot).

At first read, there were 5 processes. And 5 processes displayed in the popup.
On the second one only 2 processes. So it draws only 2 (from the bottom). But 3 processes from the previous attempt leave. 

And because of this, you receive a list of 5 processes. Top 3 from the first read, and last 2 from the second read.

exelban, 2020-11-18:
Sorry for such spam)
Could you check this build?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5562681/Stats.dmg.zip)

Yoshi2889, 2020-11-19:
Thanks. I've set Stats to show 8 processes. Using your build, immediately after launch I see only 4 processes, and two of them are duplicated. They all show pid 0.

[image]

exelban, 2020-11-19:
It's in the last build?

Yoshi2889, 2020-11-19:
This is the build you gave me in your last comment.

exelban, 2020-11-19:
Check please this. If nothing changed enable please debug mode.
To do it you must run the app with the `--debug` argument. You can do this with: `/Applications/Stats.app/Contents/MacOS/Stats --debug` But run this command for this build. You will need to replace `/Applications` with this build path.
Debug mode will write some logs to the `Documents` (`log.txt`). Could you please post these logs?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5565264/Stats.dmg.zip)

Yoshi2889, 2020-11-19:
Here you go. It indeed looks like it's getting too little processes as output and doesn't overwrite the ones already in the list with the new set.

Maybe it would help if you'd wipe the widget's list before every update? Perhaps clearing `self.processes` on every processCallback call would resolve this?
https://github.com/exelban/stats/blob/58f668b81cbb53f24f8d93ca2838b79618d63663/Modules/Memory/popup.swift#L232-L249

[log.txt](https://github.com/exelban/stats/files/5565598/log.txt)

exelban, 2020-11-19:
Yeap, it will solve duplication. But the problem not there. You must see all 8 processes. Not random number)

exelban, 2020-11-19:
Hmm, it looks like `top` returns less than 8 processes. Or I make some error in the `regex`. Will check this.

Yoshi2889, 2020-11-19:
I have been trying some things locally and have at least solved the duplication issue. I don't know why I am not seeing 8 processes but at least this will make the code a bit more robust. I will open a PR.

exelban, 2020-11-19:
Thanks. I will take a look later today. Could you please post the output of `top`: 
`/usr/bin/top -l 1 -o mem -n 8 -stats pid,command,mem`. In the text format if you can.

Yoshi2889, 2020-11-19:
```
❯ top -l 1 -o mem -n 8 -stats pid,command,mem
Processes: 464 total, 3 running, 3 stuck, 458 sleeping, 1733 threads 
2020/11/19 11:21:31
Load Avg: 2.23, 3.86, 4.60 
CPU usage: 10.0% user, 23.33% sys, 66.66% idle 
SharedLibs: 269M resident, 96M data, 38M linkedit.
MemRegions: 54370 total, 1755M resident, 84M private, 627M shared.
PhysMem: 7068M used (2283M wired), 1123M unused.
VM: 2166G vsize, 2308M framework vsize, 816558(0) swapins, 1211428(0) swapouts.
Networks: packets: 2378132/1669M in, 4703060/2491M out.
Disks: 2600376/27G read, 517673/12G written.

PID   COMMAND          MEM  
7752  phpstorm         1819M
125   WindowServer     592M 
1806  backupd          227M 
0     kernel_task      116M 
8049  node             106M 
8054  node             86M  
331   Finder           83M  
359   NotificationCent 62M
```

Here you go. Note that the two `node` processes are not the same.

exelban, 2020-11-19:
I hope this one will display all processes.
[Stats.dmg.zip](https://github.com/exelban/stats/files/5567418/Stats.dmg.zip)

Yoshi2889, 2020-11-19:
I'm sorry, I completely missed your e-mail.

That version is looking much better!
[image]

Two tiny issues:
- It still shows pid 0 for every process in the tooltip.
- Icons don't seem to appear.

Was it the regex as you expected?

exelban, 2020-11-19:
Thanks for your help. Yeap, it was a regex. Even a few of them)
Your `top` output help to improve it.

This fix will be in the next release (Sunday). But you can use this build till release.
[Stats.dmg.zip](https://github.com/exelban/stats/files/5569417/Stats.dmg.zip)

closed 2020-11-22 by exelban