exelban/stats #170: Deleting wrong element while enumerating sensors

Military-Doctor, 2020-11-18:
**Describe the bug**
Here are a bug when you are trying to delete an sensor which temperature is higher than 110 Celsius.

Due to the removing during the enumerating.

**Details:**
 - Device: [MacBook Pro 16'' 2019, i7-9750H, 16G]
 - MacOS [10.15.7 (19H15)]
 - Application version [2.4.3]

**Explain**
`Modules/Sensors/readers.swift`

It's a for loop about line 44, I added a line to print the list count to see if the wrong sensors are deleted.

Then I also add one line to print out the sensors before and after the deleting.

```
        print("Sensors:", list.count);
        for (index, sensor) in list.enumerated() {
            print(index, sensor.key, sensor.name,sensor.formattedMiniValue, sensor.formattedValue);
            if let newValue = self.smc.pointee.getValue(sensor.key) {
                // Remove the temperature sensor, if SMC report more that 110 C degree.
                if sensor.type == SensorType.Temperature.rawValue && newValue > 110 {
                    print("Removing",index, sensor.key);
                    list.remove(at: index)
                    continue
                }
                
                if let idx = list.firstIndex(where: { $0.key == sensor.key }) {
                    list[idx].value = newValue
                }
            }
        }
        print("After remove", list.count);
        for(index, sensor) in list.enumerated(){
            print(index, sensor.key, sensor.name,sensor.formattedMiniValue);
            
        }
        self.list = list
```

The results are

```diff
@@Sensors: 32@@
0 Th1H Heatpipe 2 0° 0°C
1 Th2H Heatpipe 3 0° 0°C
2 TC0E CPU diode virtual 0° 0°C
3 TC0F CPU diode filtered 0° 0°C
4 TC0P CPU proximity 0° 0°C
5 TC1C CPU core 2 0° 0°C
6 TC2C CPU core 3 0° 0°C
7 TC3C CPU core 4 0° 0°C
8 TC4C CPU core 5 0° 0°C
9 TC5C CPU core 6 0° 0°C
10 TC6C CPU core 7 0° 0°C
- 11 TC7C CPU core 8 0° 0°C
@@Removing 11 TC7C@@
12 TC8C CPU core 9 0° 0°C
@@Removing 12 TC8C@@
- 13 TCGC GPU Intel Graphics 0° 0°C
14 TG0P GPU proximity 0° 0°C
15 Tm0P Mainboard 0° 0°C
16 TB1T Battery 0° 0°C
17 TW0P Airport 0° 0°C
18 VCAC CPU IA 0.0V 0.000V
19 VCSC CPU System Agent 0.0V 0.000V
20 VCTC GPU Intel Graphics 0.0V 0.000V
21 VG0C GPU 0.0V 0.000V
22 VD0R DC In 0.0V 0.000V
23 VP0R 12V rail 0.0V 0.000V
24 PCPC CPU Package 0.0W 0.00W
25 PCPT CPU Package total 0.0W 0.00W
26 PC0R CPU Computing high side 0.0W 0.00W
27 PCPG GPU Intel Graphics 0.0W 0.00W
28 PG0R GPU 0.0W 0.00W
29 PPBR Battery 0.0W 0.00W
30 PDTR DC In 0.0W 0.00W
31 PSTR System total 0.0W 0.00W
@@After remove 30@@
0 Th1H Heatpipe 2 46°
1 Th2H Heatpipe 3 51°
2 TC0E CPU diode virtual 58°
3 TC0F CPU diode filtered 61°
4 TC0P CPU proximity 53°
5 TC1C CPU core 2 61°
6 TC2C CPU core 3 60°
7 TC3C CPU core 4 60°
8 TC4C CPU core 5 59°
9 TC5C CPU core 6 58°
10 TC6C CPU core 7 57°
11 TC8C CPU core 9 0°
12 TG0P GPU proximity 49°
13 Tm0P Mainboard 52°
14 TB1T Battery 36°
15 TW0P Airport 49°
16 VCAC CPU IA 1.1V
17 VCSC CPU System Agent 0.0V
18 VCTC GPU Intel Graphics 0.0V
19 VG0C GPU 0.0V
20 VD0R DC In 20.1V
21 VP0R 12V rail 12.6V
22 PCPC CPU Package 16.6W
23 PCPT CPU Package total 19.2W
24 PC0R CPU Computing high side 24.8W
25 PCPG GPU Intel Graphics 0.2W
26 PG0R GPU 0.1W
27 PPBR Battery 0.2W
28 PDTR DC In 55.4W
29 PSTR System total 27.7W
```

As you can see that it did delete 2 sensors, but the Intel Graphics sensor was deleted.

The easiest way to fix this bug is to change

```swift
for (index, sensor) in list.enumerated() { ...
```

to 

```swift
for (index, sensor) in list.enumerated().reversed() { ...
```

exelban, 2020-11-18:
Hi. Thanks for such a nice description)
I will take a look at this bug tomorrow. 

Do you maybe know why it's happening? Looks strange that just reversing the list helps.

Military-Doctor, 2020-11-19:
## Let me start with the conclusion:

**When using the reverse iterator, deleting an element will not affect the index of the previous elements!**

## Here are some details that can be ignored:

I don't know how Swift works. 

But I think it's similar to the `Reverse Iterator` in C++.

After looking up some details, I think this bug is because swift generates a `temporary list` when you call `list.enumerated()` 

for example, the `list` is:
```swift
[A,B,C,D,E]
```

the generated `temporary list` after calling `list.enumerated()` is:
````swift
[
(0, A)
(1, B)
(2, C)
(3, D)
(4, E)
]
````

Then, if you are trying to delete the `B` in the `list`, say `list.remove(at: 1)`

The `list` will becomes to 

```swift
[A,C,D,E]
```

However, the `temporary list` still remains:

```swift
[
(0, A)
(1, B)
(2, C)
(3, D)
(4, E)
]
```

Then, if you want to delete the `D` in the `list`, say `list.remove(at: 3)` 
(index is 3 is because you are using the index in the `temporary list`)

You actually delete the `E`!

Because the index of the list has changed since the last deletion, you HAVE TO regenerate a `temporary list`, or let the index minus 1.

**However, when you are using ```list.enumerated().reversed()```**

**Deleting the element will not affect the index of the previous elements!**

Such an interesting details, haha.

Icarus-Xu, 2020-11-19:
> ## Let me start with the conclusion:
> **When using the reverse iterator, deleting an element will not affect the index of the previous elements!**
>
> ## Here are some details that can be ignored:
> I [...]

exactly

exelban, 2020-11-19:
@Military-Doctor ohh. Now it makes a lot of sense for me. 
Thank you very much for such a nice explanation)

exelban, 2020-11-19:
#171 merged. This fix will be available in the next release (Sunday).

Military-Doctor, 2020-11-19:
> #171 merged. This fix will be available in the next release (Sunday).

Thanks~

closed 2020-11-19 by Military-Doctor

finnrobinanders, 2020-12-07:
Just updated to the latest version and now there _are_ 6 CPU core temps showing however they are numbered 2-7. Are they the correct core temps but simply mis-numbered? The 264˚F sensor is now gone.
Also, there is a new font for the temperature sensors and fan speeds that does not match the rest of the stats.
[image]