exelban/stats #172: [Bug] wrong cpu core temperature

Icarus-Xu, 2020-11-19:
**Describe the bug**
i7-9750h has 6 cores but the display shows core #2-7 (should be #0-5 i think), and there is another core #9 and its temperature is 129 degrees, i guess it's not correct

**Details:**
 - Device: [Macbook Pro 16inch late 2019]
 - MacOS [11.1 Beta 20C5048k]
 - Application version [2.4.3]

Icarus-Xu, 2020-11-19:
[image]

Military-Doctor, 2020-11-19:
Hello, I have the same bug.

Check the issue #170 for more details.

Icarus-Xu, 2020-11-19:
@Military-Doctor thanks for replying, i will check them out

exelban, 2020-11-19:
Duplicate of #150

closed 2020-11-19 by exelban