exelban/stats #173: [Bug] radeon 5300m is not showing up

Icarus-Xu, 2020-11-19:
**Describe the bug**
cannot monitor amd graphic card, the app only shows intel uhd 630

[image]

**Details:**
 - Device: [Macbook Pro 16inch late 2019]
 - MacOS [11.1 Beta 20C5048k]
 - Application version [2.4.3]

Icarus-Xu, 2020-11-19:
[image]

as you can see in the "gpu to show" list, there is only intel 630

Military-Doctor, 2020-11-19:
I am the same model as you and have the same problem.

I read the code and found that the program can get two graphics cards properly. 

I am still investigating the specific reason about why the AMD graphics card cannot be displayed.

Military-Doctor, 2020-11-19:
Maybe you can disable the `Automatic graphics switching` to let your AMD 5300M enabled.

After the graphics card working mode is set to `Discrete Only`, the AMD card will show up.

[image]

[image]

Icarus-Xu, 2020-11-19:
thanks for replying, but after i turned off auto graphic switching, this "gpu to show" list is simply empty :( I'm not sure what's wrong.

Icarus-Xu, 2020-11-19:
@Military-Doctor

Icarus-Xu, 2020-11-19:
ah ok, I got it right. but still the temperature is not correct on radeon, not sure if it has a sensor and can read its value

exelban, 2020-11-19:
Hi. Thanks for reporting.
I suppose the temperature is not available, because of the driver not providing it.
Could you please post `ioreg -r -d 1 -w 0 -c "IOAccelerator" > output.txt`?

Icarus-Xu, 2020-11-19:
```
+-o AMDRadeonX6000_AMDNavi14GraphicsAccelerator  <class AMDRadeonX6000_AMDNavi14GraphicsAccelerator, id 0x100000727, registered, matched, active, busy 0 (209 ms), retain 113>
    {
      "AccelCaps" = 15
      "IODVDBundleName" = "AMDRadeonVADriver2"
      "IOGVAHEVCEncode" = "1"
      "ATY,intrev" = 21
      "AccelNativeDMARowByteAlignment" = 256
      "InternalStatisticsAccm" = {}
      "MetalStatisticsName" = "GFX10Statistics"
      "GPURawCounterBundleName" = "AMDRawCounterPlugin"
      "GPUDCCDisplayable" = Yes
      "IOGeneralInterest" = "IOCommand is not serializable"
      "InternalStatistics" = {}
      "MetalPluginClassName" = "AMDMTLGFX10Device"
      "GPURawCounterPluginClassName" = "AMDRawCounterSourceGroup"
      "GpuDebugPolicy" = 0
      "AAPL,OpenCLdisabled" = Yes
      "AGCInfo" = {"fBusyCount"=0,"poweredOffByAGC"=Yes,"fSubmissionsSinceLastCheck"=0,"fLastSubmissionPID"=1482}
      "dpm" = 1
      "MetalPluginName" = "AMDRadeonX6000MTLDriver"
      "PerformanceStatisticsAccum" = {"orphanedReusableVidMemoryHitRate"=87,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=0,"dataBufferCount"=0,"oolTexturePageInBytes"=0,"textureCount"=1563,"iosurfaceTextureCreationCount"=0,"surfaceReadLockIdleWaitTime"=0,"freeDataBufferWaitTime"=0,"orphanedReusableSysMemoryBytes"=0,"ioSurfaceReadOutBytes"=0,"agprefTextureCreationCount"=0,"orphanedReusableSysMemoryCount"=0,"finishAll2DWaitTime"=0,"orphanedNonReusableSysMemoryCount"=0,"ioSurfacePageOutBytes"=0,"surfaceBufferReadOutBytes"=0,"surfaceTextureCreationCount"=0,"textureReadOutBytes"=0,"gartMapInBytesPerSample"=0,"swapBytesPerSample"=0,"dataBytesPerSample"=0,"surfaceCount"=117,"clientGLWaitTime"=0,"surfaceCopyInWaitTime"=0,"clientSharedAllocatedBytes"=1224704,"surfaceCopyOutWaitTime"=0,"gartMapOutBytesPerSample"=0,"ioSurfacePageInBytes"=0,"gartCacheBytes"=33554432,"volatileSurfaceCount"=0,"vramEvictionWaitTime"=0,"oolTextureCreationBytes"=0,"freeToAllocGPUAddressWaitTime"=0,"oolTextureCreationCount"=0,"orphanedReusableVidMemoryCount"=0,"orphanedReusableVidMemoryBytes"=0,"context2DCount"=5,"inUseSysMemoryBytes"=3936256,"bufferSwapCount"=0,"orphanedNonReusableVidMemoryBytes"=0,"contextGLCount"=4,"finish2DWaitTime"=0,"gartSizeBytes"=6442450944,"recoveryCount"=0,"stdTexturePageInBytes"=0,"agprefTextureCreationBytes"=0,"gartUsedBytes"=3936256,"surfaceWriteLockIdleWaitTime"=0,"surfaceBufferPageInBytes"=0,"finishGLWaitTime"=0,"iosurfaceTextureCreationBytes"=0,"finishCLWaitTime"=0,"finishVideoWaitTime"=0,"contextCLCount"=0,"freeSurfaceBackingWaitTime"=0,"swapCompleteGLWaitTime"=0,"orphanedNonReusableSysMemoryBytes"=0,"inUseVidMemoryBytes"=0,"agpTextureCreationBytes"=0,"stdTextureCreationBytes"=0,"agpTextureCreationCount"=0,"stdTextureCreationCount"=0,"surfaceSetShapeIdleWaitTime"=0,"gartFreeBytes"=6438514688,"textureVolunteerUnloadBytes"=0,"bufferFlipCount"=0,"hardwareSubmitWaitTime"=0,"swapComplete2DWaitTime"=0,"surfaceBufferPageOutBytes"=0,"surfaceBufferTextureCreationCount"=0,"hardwareWaitTime"=0,"contextVideoCount"=0,"freeSurfaceSwapBufferWaitTime"=0,"orphanedNonReusableVidMemoryCount"=0}
      "PerformanceStatistics" = {"orphanedReusableVidMemoryHitRate"=87,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=0,"dataBufferCount"=0,"oolTexturePageInBytes"=0,"textureCount"=1563,"iosurfaceTextureCreationCount"=0,"surfaceReadLockIdleWaitTime"=0,"freeDataBufferWaitTime"=0,"orphanedReusableSysMemoryBytes"=0,"ioSurfaceReadOutBytes"=0,"agprefTextureCreationCount"=0,"orphanedReusableSysMemoryCount"=0,"finishAll2DWaitTime"=0,"orphanedNonReusableSysMemoryCount"=0,"ioSurfacePageOutBytes"=0,"surfaceBufferReadOutBytes"=0,"surfaceTextureCreationCount"=0,"textureReadOutBytes"=0,"gartMapInBytesPerSample"=0,"swapBytesPerSample"=0,"dataBytesPerSample"=0,"surfaceCount"=117,"clientGLWaitTime"=0,"surfaceCopyInWaitTime"=0,"clientSharedAllocatedBytes"=1224704,"surfaceCopyOutWaitTime"=0,"gartMapOutBytesPerSample"=0,"ioSurfacePageInBytes"=0,"gartCacheBytes"=33554432,"volatileSurfaceCount"=0,"vramEvictionWaitTime"=0,"oolTextureCreationBytes"=0,"freeToAllocGPUAddressWaitTime"=0,"oolTextureCreationCount"=0,"orphanedReusableVidMemoryCount"=0,"orphanedReusableVidMemoryBytes"=0,"context2DCount"=5,"inUseSysMemoryBytes"=3936256,"bufferSwapCount"=0,"orphanedNonReusableVidMemoryBytes"=0,"contextGLCount"=4,"finish2DWaitTime"=0,"gartSizeBytes"=6442450944,"recoveryCount"=0,"stdTexturePageInBytes"=0,"agprefTextureCreationBytes"=0,"gartUsedBytes"=3936256,"surfaceWriteLockIdleWaitTime"=0,"surfaceBufferPageInBytes"=0,"finishGLWaitTime"=0,"iosurfaceTextureCreationBytes"=0,"finishCLWaitTime"=0,"finishVideoWaitTime"=0,"contextCLCount"=0,"freeSurfaceBackingWaitTime"=0,"swapCompleteGLWaitTime"=0,"orphanedNonReusableSysMemoryBytes"=0,"inUseVidMemoryBytes"=0,"agpTextureCreationBytes"=0,"stdTextureCreationBytes"=0,"agpTextureCreationCount"=0,"stdTextureCreationCount"=0,"surfaceSetShapeIdleWaitTime"=0,"gartFreeBytes"=6438514688,"textureVolunteerUnloadBytes"=0,"bufferFlipCount"=0,"hardwareSubmitWaitTime"=0,"swapComplete2DWaitTime"=0,"surfaceBufferPageOutBytes"=0,"surfaceBufferTextureCreationCount"=0,"hardwareWaitTime"=0,"contextVideoCount"=0,"freeSurfaceSwapBufferWaitTime"=0,"orphanedNonReusableVidMemoryCount"=0}
      "IOGLBundleName" = "AMDRadeonX6000GLDriver"
      "IOPropertyMatch" = {"LoadAccelerator"=Yes}
      "IOKitDebug" = 0
      "IOSourceVersion" = "0.0.0.0.0"
[... cut ...]
      "AccelCaps" = 15
      "IOAccelDisplayPipeCapabilities" = {"DisplayPipeSupported"=Yes,"TransactionsSupported"=Yes}
      "IOGVAHEVCEncodeCapabilities" = {"VTPerProfileDetails"={"1"={"VTMaxEncodeLevel"=156},"2"={"VTMaxEncodeLevel"=156}},"VTQualityRating"=80,"VTRating"=100,"VTSupportedProfileArray"=(1,2)}
      "IOProbeScore" = 1000
      "SurfaceList" = ({"surfaceType"="ctx","width"=64,"height"=64,"pid"=2930},{"surfaceType"="ctx","width"=64,"height"=64,"pid"=2930})
    }
    

```

@exelban

Icarus-Xu, 2020-11-19:
yep nothing looks like "tempreture"

exelban, 2020-11-19:
Yeap. Unfortunately, I don't see any temperature(
And do you have the temperature on Intel graphics? Because it must be unavailable too according to this output.

Icarus-Xu, 2020-11-19:
@exelban that would be strange, I do have temp on intel graphics, and it gets updated periodically

Icarus-Xu, 2020-11-19:
[image]

exelban, 2020-11-19:
So maybe you run this command when IG was disabled. Because recording to the output. There is no information about IG temperature.

Unfortunately, I don't find any other way to obtain GPU usage/temperature from macOS. Only nvidia/interl/amd drivers can provide this info. And if they do not provide this info, I cannot obtain it(

Military-Doctor, 2020-11-19:
I have the same problem.

exelban, 2020-11-19:
@Military-Doctor you can help do debug it. Just add a few prints to the GPU reader (I suppose you can run the app locally)).

exelban, 2020-11-29:
Closed due inactivity.

closed 2020-11-29 by exelban