exelban/stats #175: RW speed not shown correctly in the icon

HartwigHarder, 2020-11-19:
While my system is writing and reading intensively, the corresponding stat shows 0/0 for read and write access.

exelban, 2020-11-19:
Hi.

How many disks do you have in the system?

HartwigHarder, 2020-11-19:
one ssd

exelban, 2020-11-19:
So in the popup view there are no activities too?

exelban, 2020-11-19:
Could you please provide the mac model and the version of macOS?

HartwigHarder, 2020-11-19:
activities are shown in the system activity monitor,
MacBook Pro 13", 2017
Big Sur 11.0.1

exelban, 2020-11-19:
Would you like to help by running a debug version of the app?

HartwigHarder, 2020-11-19:
yes

exelban, 2020-11-19:
Thanks.

You must run the app with the `--debug` argument. You can do this with: `/Applications/Stats.app/Contents/MacOS/Stats --debug` But run this command for this build. You will need to replace `/Applications` with this build path.
Debug mode will write some logs to the `Documents` (`log.txt`). Could you please post these logs?

[Stats.dmg.zip](https://github.com/exelban/stats/files/5568873/Stats.dmg.zip)

HartwigHarder, 2020-11-20:
----------- 2020-11-20 08:57:44 +0000 ------------
Found 1 display devices
------------ CapacityReader.read() ------------
driveStats
Disk parent: 74755
[DISK] Callback: 1 disks
RAM: found 8 top processes
------------ CapacityReader.read() ------------
driveStats
[DISK] Callback: 1 disks
------------ CapacityReader.read() ------------
driveStats
[DISK] Callback: 1 disks

HartwigHarder, 2020-11-20:
I just copied the first couple of lines as the last lines are repetitive

exelban, 2020-11-20:
Could you please do the same with this one?
[Stats.dmg.zip](https://github.com/exelban/stats/files/5572610/Stats.dmg.zip)

HartwigHarder, 2020-11-20:
------------ 2020-11-20 10:39:11 +0000 ------------
driveStats
get props
{
    "BSD Major" = 1;
    "BSD Minor" = 3;
    "BSD Name" = disk1;
    "BSD Unit" = 1;
    Content = "EF57347C-0000-11AA-AA11-00306543ECAC";
    "Content Hint" = "EF57347C-0000-11AA-AA11-00306543ECAC";
    Ejectable = 0;
    EncryptionBlockSize = 4096;
    IOBusyInterest = "IOCommand is not serializable";
    IOGeneralInterest = "IOCommand is not serializable";
    IOStorageFeatures =     {
        Barrier = 1;
        Priority = 1;
        Unmap = 1;
    };
    Leaf = 0;
    "Logical Block Size" = 4096;
    Open = 1;
    "Physical Block Size" = 4096;
    "Preferred Block Size" = 4096;
    Removable = 0;
    Size = 1000240963584;
    UUID = "9AEDA00B-BE24-4C1D-8E83-5CD0406A0EF6";
    Whole = 1;
    Writable = 1;
}

exelban, 2020-11-20:
Thanks. Unfortunately, the app cannot obtain disk statistics. It looks like this disk does not report it.
Did you use the default mac disk? Or it was replaced?

HartwigHarder, 2020-11-20:
It is the original disk
[image]

HartwigHarder, 2020-11-20:
in the activity monitor it does show the access

exelban, 2020-11-20:
Nice, Apple has a much better library)
Also, I suppose they use internal drivers to fetch the stats.
I will be very glad if someone will give a better idea of how I can fetch disk stats)

HartwigHarder, 2020-11-20:
when I use iostat

HartwigHarder, 2020-11-20:
[image]

HartwigHarder, 2020-11-20:
maybe that gives a trace

exelban, 2020-11-20:
I'm trying to avoid calling 3d party apps in the main readers. It increases cpu usage. And this read must be triggered each second (or another interval). So this way cannot be applied(

HartwigHarder, 2020-11-20:
yes, didn't meant you should use iostat, but maybe looking into the way how iostat is doing it will give some information

exelban, 2020-11-20:
But it's really very strange that in the last logs there are no stats. They normally must be. Maybe there is some problem with detecting the parent partition of the disk. But it requires more deeply debugging. If you are ready to help with this. I would be very appreciated. I suppose it will take a few days and a lot of iteration(

HartwigHarder, 2020-11-20:
sure, my fun as well :)
first suspicion had been that paragon NTFS would interfere, but I disabled it and still no change

exelban, 2020-11-20:
Thanks. I will try to add some logs which have sense. And post a debug build.
Nope, NTFS cannot make a problem.

exelban, 2020-11-20:
Small question. Do you use APFS (I'm not sure if Big Sur can be installed on another fs, but still)?

HartwigHarder, 2020-11-20:
yes, using apfs

exelban, 2020-11-20:
Could you please post ouput of `diskutil apfs list`?
Please replace sensitive data from the output with some symbols.

exelban, 2020-11-20:
I'm sure for 75% that the problem is in the code where I'm trying to detect physical disk of partition.

HartwigHarder, 2020-11-20:
APFS Container (1 found)
|
+-- Container disk1 9AEDA00B-BE24-4C1D-8E83-5CD0406A0EF6
    ====================================================
    APFS Container Reference:     disk1
    Size (Capacity Ceiling):      1000240963584 B (1.0 TB)
    Capacity In Use By Volumes:   947479404544 B (947.5 GB) (94.7% used)
    Capacity Not Allocated:       52761559040 B (52.8 GB) (5.3% free)
    |
    +-< Physical Store disk0s2 CB348DC0-BCEF-4E55-812E-C8DCA1BECEB5
    |   -----------------------------------------------------------
    |   APFS Physical Store Disk:   disk0s2
    |   Size:                       1000240963584 B (1.0 TB)
    |
    +-> Volume disk1s1 79850DFE-A93C-34DA-AF64-CE77F9264486
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s1 (Data)
    |   Name:                      Macintosh HD - Daten (Case-insensitive)
    |   Mount Point:               /System/Volumes/Data
    |   Capacity Consumed:         928165392384 B (928.2 GB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk1s2 4E33F84D-F3DD-43E8-9C71-D6ADFC530BA7
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s2 (Preboot)
    |   Name:                      Preboot (Case-insensitive)
    |   Mount Point:               /System/Volumes/Preboot
    |   Capacity Consumed:         288112640 B (288.1 MB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk1s3 22A35D5F-02E6-4F4D-8ADE-318786BFC3CB
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s3 (Recovery)
    |   Name:                      Recovery (Case-insensitive)
    |   Mount Point:               Not Mounted
    |   Capacity Consumed:         655523840 B (655.5 MB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk1s4 11829515-2C9D-4DCC-89A2-22C762BA5A06
    |   ---------------------------------------------------
    |   APFS Volume Disk (Role):   disk1s4 (VM)
    |   Name:                      VM (Case-insensitive)
    |   Mount Point:               /System/Volumes/VM
    |   Capacity Consumed:         3222376448 B (3.2 GB)
    |   Sealed:                    No
    |   FileVault:                 No
    |
    +-> Volume disk1s5 93ED18E4-1916-4F4E-80AE-4C9C79F8C677
        ---------------------------------------------------
        APFS Volume Disk (Role):   disk1s5 (System)
        Name:                      Macintosh HD (Case-insensitive)
        Mount Point:               Not Mounted
        Capacity Consumed:         14939553792 B (14.9 GB)
        Sealed:                    Broken
        FileVault:                 No
        |
        Snapshot:                  72304C91-5930-4D30-8D79-7EE88CB02FC1
        Snapshot Disk:             disk1s5s1
        Snapshot Mount Point:      /
        Snapshot Sealed:           yes

HartwigHarder, 2020-11-20:
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *1.0 TB     disk0
   1:                        EFI ⁨EFI⁩                     314.6 MB   disk0s1
   2:                 Apple_APFS ⁨Container disk1⁩         1.0 TB     disk0s2

exelban, 2020-11-20:
Thanks, I figured out where is the problem.

HartwigHarder, 2020-11-20:
:D ??

exelban, 2020-11-20:
In the beginning, I'm receiving the available disks (but it's only mounted partitions). So I must find a whole drive going to the top of the partition hierarchy. Normally it takes 2-3 steps. But `apfs` has different levels of hierarchy. I have a Catalina installed, and from a mounted partition (`Machintosh HD`) to the physical disk app must take a 3 step up. But I check the Big Sur, and the same `Macintosh HD` partition is on 4 levels down. So it requires an additional step up to found physical drive.

I must find parent drive because only on this level I can collect the write/read stats.

Check this build please (just run, without logs).
[Stats.dmg.zip](https://github.com/exelban/stats/files/5574583/Stats.dmg.zip)

exelban, 2020-11-20:
I'm not sure if it's a new behavior of macOS or just a new apfs driver. But on Catalina and Big Sur I have different levels of hierarchy)

HartwigHarder, 2020-11-20:
Works :)

exelban, 2020-11-20:
Ok, the fix will be available in the next release (on Sunday). But you can use the last build till then.
Thanks for the help.

closed 2020-11-22 by exelban