exelban/stats #177: Disk Popup not shown

wernerfred, 2020-11-20:
Started my Mac today and found the popup screen of the disk stats not showing anything:
[image]

**Details:**
 - Device: [Macbook Pro 2019]
 - MacOS [11.0.1]
 - Application version [2.4.3]

Don't know what else to provide, any advise and i will deliver the requested data :)

exelban, 2020-11-20:
Hi. Thanks for reporting.

I have noticed it on macOS Big Sur.
It's already [fixed](https://github.com/exelban/stats/commit/eccdf6e9a7392926ff8bdf4f95d4158463dff6e1) and will be available in the next release.

exelban, 2020-11-20:
Temporary fix: restart the app. There is a small racing, and from time to time it's not adding to the view(

closed 2020-11-20 by exelban

wernerfred, 2020-11-20:
You are insane! Thumbs up for your work 👍🏻