exelban/stats #179: Desktop switches when clicking on top bar icon

th3dic3m4n, 2020-11-20:
Hi there,

When clicking on a top bar element in secondary desktops (say Desktop 3), the screen moves to Desktop 1 and the detailed Stats dropdown is shown, instead of just showing it directly in Desktop 3.

If needed can screencapture it, but I think it's quite understandable.

Cheers & thanks for this great tool!
Frank

 - Macbook Pro 2016 / MacOS 10.15.7
 - Stats v2.4.3

exelban, 2020-11-20:
Hi. Thanks for the feedback.

Do you mean multi-monitor setup? Or macOS desktops?

th3dic3m4n, 2020-11-20:
Hi, 

Sorry for not making it clear, I mean macOS desktops.

exelban, 2020-11-20:
Hmm, so it's always returning to the first desktop on the list and showing a popup on this desktop?
Strange behavior. I'm interested why I cannot reproduce it locally(

exelban, 2020-11-20:
Normally I'm using a few desktops on the macbook. Never noticing something like this. Maybe you have some specific configuration of macOS?

th3dic3m4n, 2020-11-20:
Great, found out what it is. it happens only between Desktop where I have a full screen app (like Chrome). If I have Desktop 1 + 2 with no full screen apps, then it works as expected.

th3dic3m4n, 2020-11-20:
So meaning, Desktop 3 has Chrome Full Screen, there the top bar click moves to Desktop 1, but Desktop 2 without full screen apps, works perfectly

exelban, 2020-11-20:
Still strange. So it moves to Desktop 1 only when you have a few full screen desktops?
Or only when the 3 one is full screen Chrome and another doesn't matter?

th3dic3m4n, 2020-11-20:
Here a screen cap of the Desktops http://snpy.in/lc5o5K

Desktop 1: Finder, no full screen
Desktop 2: Apps, no full screen
Desktop 3 ("Desktop switches when"): Chrome full screen

If I click on Top bar Stats icons (for example CPU) when in Desktop 3 --> it switches to Desktop 1 and shows the CPU dropdown there and not in Desktop 3.

If I do the same in Desktop 2, it stays in Desktop 2.

exelban, 2020-11-20:
I still cannot reproduce it with the same desktop configurations. 
It may be some Chrome bug.  Could you try to open on Desktop 3 some other app in full screen and check if still go back to Desktop 1?

th3dic3m4n, 2020-11-20:
Happening with any other full screen app, iTunes, Terminal, ecc.
And it's not happening with any other native or 3rd party apps on the top bar

exelban, 2020-11-20:
I will take a look at what can cause this move back. But not promising anything.

th3dic3m4n, 2020-11-20:
I will also do a clean restart and see if it persists, thanks & keep in touch!

exelban, 2020-11-29:
Hi. I trying to recreate this bug. But without any successes(
I suppose that it can be something in desktop configuration or macOS itself.

I will close the issue. But if it will repeat or someone will have the same bug, please reopen this issue.

closed 2020-11-29 by exelban