exelban/stats #180: Glitch when using multiple SSDs

richardtop, 2020-11-21:
**Describe the bug**
When using multiple SSDs, after disconnecting an external SSD the widget shows two Macintosh HD with one of them being 100% full.
[image]

After restarting, the disk list is empty:
[image]

**Details:**
 - Device: [e.g. Macbook Pro 16]
 - MacOS [e.g. 11]
 - Application version [e.g. 2.1.11]

exelban, 2020-11-22:
Hi. Thanks for reporting. I must take a looks at why it's showing 100% when reconnecting.
Empty popup view described [here](https://github.com/exelban/stats/issues/177).

richardtop, 2020-11-22:
Is it fixed in the latest release?

richardtop, 2020-11-22:
Also, how about a duplicate disk?
[image]

exelban, 2020-11-22:
Duplicating disks is something that must be fixed. Only the empty view fixed in the last version.

exelban, 2020-11-29:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.5) release.

closed 2020-11-29 by exelban