exelban/stats #181: Apple Silicon support

Wilkolicious, 2020-11-22:
**Describe the bug**
Numerous bugs or unsupported modules:

## Info/meta
- Unknown version
- Processor looks raw
- Graphics
[image]

## GPU
- No GPU selectable
- GPU stats therefore "unknown"

[image]
[image]

## RAM
- Details don't seem to match?
[image]

## Running under Rosetta 2 (not compiled for aarch64/arm64)
- See architecture column - `Intel` indicates Rosetta 2 translation (`x86-64` -> `aarc64`), `Apple` indicates native compilation
[image]

**Details:**
 - Device: Macbook Pro 13", Apple Silicon M1, 8GB
 - MacOS 11 (Big Sur)
 - Application version: 2.4.3

Appreciate any help!

thebitstick, 2020-11-22:
I have an M1 Mac Mini with 16GB. I'd be willing to help out with testing.

thebitstick, 2020-11-22:
I tested compilation of commit aff3406, and it failed on trying to compile `CPU.c`
[image]

wfrank21, 2020-11-22:
@thebitstick That is no surprise because stats uses the Intel Power Gadget Library, which, for obvious reasons, you can't use on Apple Silicon.

exelban, 2020-11-22:
Hi. Thanks for the feedback. 

I want to make it works on apple silicon. But I think it cannot be done till I will have some mac with it. Because there is a lot of difference between intel and apple silicon. At least ram, gpu, and some low level methods.
I totally disabled arm compiling (because of IntelPowerGadget). I'm not sure for now how it can be fixed.

If you running this app using Rosetta, it will not works correctly. Because this is a monitoring app. And Rosetta is a king of virtualization or simulation. And it cannot be used for monitoring the real system.

exelban, 2020-11-22:
Could someone check if this one works natively on Apple Silicon?
[Stats.dmg.zip](https://github.com/exelban/stats/files/5579519/Stats.dmg.zip)

thebitstick, 2020-11-22:
[image]
Compared to before, the processor is different, although it can't detect my Mac model.
[image]
Sensors doesn't seem to work anymore.
[image]
[image]
Not sure where GPU is getting its data from.
[image]
RAM seems accurate.

exelban, 2020-11-22:
@thebitstick Thanks.
Model is not defined because it's not in the predefined list.
Hhmm, sensors must work. Because there are not many differences.
Graphics is unknown because there is also a predefined list of vendors (and Apple was not a vendor of GPU before apple silicon come).

exelban, 2020-11-23:
So.

To build a universal app (intel + arm) IntelPowerGadget must be removed from the app. It means that app loose the CPU frequency monitor. I can agree to remove this option to make the app useful for more users (with M1). At least till I found a solution to how the app can be built with IPW framework. 

The main problem is that IPW framework is not a universal library. It's compiled only for the Intel platform. And according to [this](https://developer.apple.com/documentation/xcode/porting_your_macos_apps_to_apple_silicon): `You cannot produce a universal version of your binary without universal versions of all linked libraries.`. I'm not sure that Intel will recompile the framework to a universal version (I think it's not possible at all).

So I think in the next version I will remove the IPG from the app. It allows to compile a universal version of the app.

exelban, 2020-11-23:
With sensors, there is another problem. There is no list of available sensors or something like this at all.
I would be very appreciated if someone would like to help to obtain a list of available sensor keys from a mac with M1. It will help to determine what sensors the new mac has and which keys I must take for different values (like CPU temperature).

thebitstick, 2020-11-23:
Sensors.. seem impossible.
I tried `sysctl`, and `powermetrics` commands, but I am unable to gather any data from sensors and fans at all.
I tried [this project](https://github.com/lavoiesl/osx-cpu-temp), but it didn't seem to work either.
I think it's dead jim.

exelban, 2020-11-23:
Try this:
- `git clone https://github.com/hholtmann/smcFanControl`
- `cd smcFanControl && cd smc-command`
- `make`
- `./smc -l`

thebitstick, 2020-11-23:
```
smc-command ) ./smc -l
  #KEY  [ui32]  863 (bytes 00 00 03 5f)
  AC-F  [ui8 ]  0 (bytes 00)
  AC-I  [ui16]  19 (bytes 00 13)
  AC-M  [hex_]  (bytes 01 01)
  AC-N  [ui8 ]  2 (bytes 02)
  AC-P  [ui8 ]  0 (bytes 00)
  AC-Q  [ui8 ]  0 (bytes 00)
  AC-R  [ui8 ]  0 (bytes 00)
  AC-S  [flag]  (bytes 01)
  AC-U  [flag]  (bytes 00)
  ACKG  [hex_]  (bytes 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 e0 76 4b 04 03 00 00 00 40 78 4b 04 03 00 00 00 36 5f 00 00 01 00 00 00 41 43 4b 47 00 00 00 00 76 00 00 00 68 65 78 5f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00)
  ACKL  [hex_]  (bytes 00)
  ACKR  [hex_]  (bytes 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 e0 76 4b 04 03 00 00 00 40 78 4b 04 03 00 00 00 36 5f 00 00 01 00 00 00 41 43 4b 52 00 00 00 00 6f 00 00 00 68 65 78 5f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
  ACKS  [hex_]  (bytes 00 00)
  ACSt  [hex_]  (bytes 07 00 00 00)
  AOPc  [hex_]  (bytes 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 e0 76 4b 04 03 00 00 00 40 78 4b 04 03 00 00 00 36 5f 00 00 01 00 00 00 41)
  AOPe  [ui8 ]  0 (bytes 00)
  AOPs  [ui16]  0 (bytes 00 00)
  AUPO  [flag]  (bytes 00)
  CLBT  [ui64]  (bytes 2f 8f f5 81 ad 02 00 00)
  CLK!  [ui8 ]  0 (bytes 00)
  CLKH  [{clh]  (bytes 40 19 01 00 80 70 00 00)
  CLKS  [fp1f]  0.79803 (bytes 66 26)
  CLKT  [ui32]  1110704128 (bytes 42 34 00 00)
  CLKU  [ui64]  (bytes 68 72 f3 64 c9 02 00 00)
  CLSD  [ui64]  (bytes 22 1d fe 7f ad 02 00 00)
  CLSP  [ui64]  (bytes 71 a3 eb ac c6 02 00 00)
  CLWK  [ui64]  (bytes b8 92 e8 d3 c6 02 00 00)
  D1CA  [ui16]  512 (bytes 02 00)
  D1CD  [ui16]  256 (bytes 01 00)
  D1Ca  [ui8 ]  1 (bytes 01)
  D1Cb  [ui8 ]  0 (bytes 00)
  D1Cc  [ui16]  0 (bytes 00 00)
  D1Ci  [ui16]  0 (bytes 00 00)
  D1Cm  [ui16]  1 (bytes 00 01)
  D1Cn  [ui16]  0 (bytes 00 00)
  D1Cw  [ui16]  0 (bytes 00 00)
  D1DA  [hex_]  (bytes 01 7e)
  D1DI  [ui8 ]  3 (bytes 03)
  D1DP  [hex_]  (bytes 00 00 00 00 44 44 00 00)
  D1DR  [flag]  (bytes 01)
  D1EC  [ui8 ]  0 (bytes 00)
  D1FV  [hex_]  (bytes 00 20 77 00)
  D1JA  [ui32]  2553937920 (bytes 98 3a 00 00)
  D1JB  [ui32]  2553937920 (bytes 98 3a 00 00)
  D1JD  [{jst]  (bytes 4c 1d 00 00 4c 1d 00 00 98 3a 98 3a 00 00 00 00 98 3a 00 00 01)
  D1JR  [si32]  (bytes 4c 1d 00 00)
  D1JS  [si32]  (bytes 4c 1d 00 00)
  D1St  [hex_]  (bytes 03 00)
  D2CA  [ui16]  256 (bytes 01 00)
  D2CD  [ui16]  0 (bytes 00 00)
  D2Ca  [ui8 ]  0 (bytes 00)
  D2Cb  [ui8 ]  0 (bytes 00)
  D2Cc  [ui16]  0 (bytes 00 00)
  D2Ci  [ui16]  0 (bytes 00 00)
  D2Cm  [ui16]  0 (bytes 00 00)
  D2Cn  [ui16]  0 (bytes 00 00)
  D2Cw  [ui16]  0 (bytes 00 00)
  D2DA  [hex_]  (bytes 01 70)
  D2DI  [ui8 ]  3 (bytes 03)
  D2DP  [hex_]  (bytes 00 00 ff 01 aa a4 00 00)
  D2DR  [flag]  (bytes 01)
  D2EC  [ui8 ]  0 (bytes 00)
  D2FV  [hex_]  (bytes 00 20 77 00)
  D2JA  [ui32]  1276968960 (bytes 4c 1d 00 00)
  D2JB  [ui32]  1276968960 (bytes 4c 1d 00 00)
  D2JD  [{jst]  (bytes 4c 1d 00 00 4c 1d 00 00 4c 1d 4c 1d 00 00 00 00 4c 1d 00 00 03)
  D2JR  [si32]  (bytes 4c 1d 00 00)
[... 773 lines cut ...]
  uuFt  [flt ]  200 (bytes 00 00 48 43)
  uuNn  [flt ]  78 (bytes 00 00 9c 42)
  uuP0  [flt ]  52 (bytes 00 00 50 42)
  uuPa  [flt ]  200 (bytes 00 00 48 43)
  uuPv  [flt ]  200 (bytes 00 00 48 43)
  uuPw  [flt ]  200 (bytes 00 00 48 43)
  uuTo  [flt ]  -100 (bytes 00 00 c8 c2)
  uuU0  [flt ]  38 (bytes 00 00 18 42)
  uuUR  [flt ]  2350 (bytes 00 e0 12 45)
  uusn  [flt ]  0 (bytes 00 00 00 00)
  voF0  [flt ]  1700 (bytes 00 80 d4 44)
  voN0  [flt ]  0 (bytes 00 00 00 00)
  voP0  [flt ]  100 (bytes 00 00 c8 42)
  voP4  [flt ]  100 (bytes 00 00 c8 42)
  voT0  [flt ]  100 (bytes 00 00 c8 42)
  voTa  [flt ]  100 (bytes 00 00 c8 42)
  voU0  [flt ]  100 (bytes 00 00 c8 42)
  w000  [si32]  (bytes 00 00 00 00)
  w001  [si32]  (bytes 00 00 00 00)
  xDPE  [flt ]  0 (bytes 30 f7 aa 3e)
  xPPT  [flt ]  65 (bytes 00 00 82 42)
  xUPT  [flt ]  15 (bytes 00 00 70 41)
```

I may not be able to decipher this, but it is a lot of zero values...

exelban, 2020-11-23:
Thanks. I will take a look at this list later today.

exelban, 2020-11-23:
A quick look at the list shows that there are totally new sensor keys(
And I think it's almost impossible to connect key -> sensors. Maybe there is some intersection between ios sensors and the m1 soc. This research will take some time. I suppose it's a good moment to create a new issue related to the Apple Silicon Sensors list.

I think at the beginning Sensors module will be just unavailable for m1 users( Till the mapping will be ready.

exelban, 2020-11-29:
New release with Apple Silicon support [available](https://github.com/exelban/stats/releases/tag/v2.4.5).

closed 2020-11-29 by exelban

Wilkolicious, 2020-11-29:
@exelban Appreciate the work!  Are there no sensor data?  There was a heatsink sensor before this update that looked sort of sane.

exelban, 2020-11-29:
@Wilkolicious there are no sensors for now. To have sensors I must have mapped sensors keys with names. I don't know how to do it(

johnalanwoods, 2020-12-04:
@exelban , it's interesting, some random apps seem to have full data from sensors, I wonder where they got their mapping/spec?

Do I understand correctly that the sensor is there, we just don't know the value to reference to read it?

exelban, 2020-12-04:
@johnalanwoods yes. The app reads all sensors from SMC. And could read the value. But, there is no mapping between sensor key and sensor name.

johnalanwoods, 2020-12-04:
Some app developers must have inside info, or indeed just guessed!

exelban, 2020-12-04:
@johnalanwoods Apple does not provide the vocabulary for the sensors. There is no good list of sensors even for intel based. 

There is only one option. Someone must help to make this list. I don't have an Apple Silicon based mac. And cannot help with this.

thebitstick, 2020-12-04:
I have access to the Sensei app which has a very rich list of sensors.
[image]
[image]
[image]

exelban, 2020-12-04:
@thebitstick thanks. There is a separate [issue](https://github.com/exelban/stats/issues/210) with this.