exelban/stats #182: Disk speed widget update interval

Tai-Zhou, 2020-11-23:
The update interval of the speed widget (4th widget) of disk is controlled by setting "update interval" (shown as "更新间隔" in the picture),
[image]
but this setting disappears when you actually choose the speed widget.
[image]

**Details:**
 - Device: [MacBook Pro 13" (Mid 2019)]
 - MacOS [11.0.1]
 - Application version [2.4.4]

exelban, 2020-11-24:
Hi. Sorry for the long response.
Yeap, it not a bug and it's not missing. Because for bandwidth (read/write) there must be no option to change the update interval. It must be always 1 second. For more information take a look [this](https://github.com/exelban/stats/issues/113) issue.

But I noticed an interesting bug. When you switch to the speed widget (read/write). This reader must automatically switch to 1 second update interval. Now it's not ready. And the user cannot see the actual read/write stats.

exelban, 2020-11-24:
Sorry, misclick)

Tai-Zhou, 2020-11-25:
I see, and thank you for your great app!

exelban, 2020-11-29:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.5) release.

closed 2020-11-29 by exelban