exelban/stats #187: this app doesn't show in menu bar when os start up.

crownzdp, 2020-11-24:


crownzdp, 2020-11-24:
os:big sur 11.01
Mac: mbp 16inch

apetronix, 2020-11-24:
Same here on Catalina 10.15.7
MBP 16 inch

Used to launch in menu bar but since updating to 2.4.4, it quits every time at start up and I have to manually launch it.

Start at login is enabled and enabled in Users & Groups > Login items

exelban, 2020-11-24:
Hi. That for reporting.
It happens only from the `2.4.4` version. Or it was also in the previous version?

crownzdp, 2020-11-24:
Not test,I'll try tomorrow

apetronix, 2020-11-24:
> Hi. That for reporting.
> It happens only from the `2.4.4` version. Or it was also in the previous version?

Wasn't in the previous version for me, just noticed this once I updated to 2.4.4

crownzdp, 2020-11-25:
2.4.3 is ok for me, just 2.4.4 version.

exelban, 2020-11-25:
Hmm, super strange. There is no difference between these apps. I mean there are no changes around the run when starting in the new version.
Thanks for checking the previous version.

crownzdp, 2020-11-25:
maybe it just crash when Mac start up

crownzdp, 2020-11-25:
Process:               Stats [566]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.4.4 (174)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [566]
User ID:               501

Date/Time:             2020-11-25 17:58:11.710 +0800
OS Version:            macOS 11.0.1 (20B50)
Report Version:        12
Bridge OS Version:     5.0.1 (18P2561)
Anonymous UUID:        AE6ED7D9-1EBE-5168-0677-8DAD3416A495

Time Awake Since Boot: 15 seconds

System Integrity Protection: enabled

Crashed Thread:        5  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [566]

Application Specific Information:
Fatal error: String index is out of bounds: file Swift/StringCharacterView.swift, line 60
 

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff2030fe7e mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff203101f0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff2043cbe7 __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2043b2ba __CFRunLoopRun + 1315
4   com.apple.CoreFoundation      	0x00007fff2043a6be CFRunLoopRunSpecific + 563
5   com.apple.HIToolbox           	0x00007fff286a6fd0 RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff286a6dcc ReceiveNextEventCommon + 709
7   com.apple.HIToolbox           	0x00007fff286a6aef _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff22c53f85 _DPSNextEvent + 883
9   com.apple.AppKit              	0x00007fff22c5274b -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1366
10  com.apple.AppKit              	0x00007fff22c44bda -[NSApplication run] + 586
11  eu.exelban.Stats              	0x000000010931415d 0x109310000 + 16733
12  libdyld.dylib                 	0x00007fff2035f631 start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff20340458 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff20340458 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff20340458 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff20340458 start_wqthread + 0

Thread 5 Crashed:: Dispatch queue: com.apple.root.default-qos
0   libswiftCore.dylib            	0x00007fff2c729064 closure #1 in closure #1 in closure #1 in _assertionFailure(_:_:file:line:flags:) + 484
1   libswiftCore.dylib            	0x00007fff2c728bb7 closure #1 in closure #1 in _assertionFailure(_:_:file:line:flags:) + 87
2   libswiftCore.dylib            	0x00007fff2c728893 closure #1 in _assertionFailure(_:_:file:line:flags:) + 99
3   libswiftCore.dylib            	0x00007fff2c728501 _assertionFailure(_:_:file:line:flags:) + 513
4   libswiftCore.dylib            	0x00007fff2c737cf7 String.index(after:) + 295
5   eu.exelban.Stats.Memory       	0x00000001094489a1 0x109442000 + 27041
6   libswiftFoundation.dylib      	0x00007fff30689bf0 partial apply for closure #1 in StringProtocol.enumerateLines(invoking:) + 32
7   libswiftFoundation.dylib      	0x00007fff30685d09 thunk for @escaping @callee_guaranteed (@guaranteed String, @unowned UnsafeMutablePointer<ObjCBool>) -> () + 73
8   com.apple.Foundation          	0x00007fff2122e2c4 -[NSString enumerateSubstringsInRange:options:usingBlock:] + 2083
[... 439 lines cut ...]
CoreUI image data                 1896K       13 
Foundation                          16K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           262.4M       66 
MALLOC guard page                   32K        6 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
SQLite page cache                   64K        1 
STACK GUARD                       56.0M        9 
Stack                             12.1M        9 
VM_ALLOCATE                        820K       34 
__CTF                               759        1 
__DATA                            13.3M      337 
__DATA_CONST                      13.9M      191 
__DATA_DIRTY                       861K      115 
__FONT_DATA                          4K        1 
__LINKEDIT                       484.7M       26 
__OBJC_RO                         61.0M        1 
__OBJC_RW                         2468K        2 
__TEXT                           223.3M      340 
__UNICODE                          588K        1 
mapped file                      221.5M       22 
shared memory                       40K        4 
===========                     =======  ======= 
TOTAL                              1.7G     1241 
TOTAL, minus reserved VM space     1.3G     1241

crownzdp, 2020-11-25:
it's crash report

exelban, 2020-11-25:
@crownzdp thanks for this research)
I will take a look at this report later today.

exelban, 2020-11-25:
@crownzdp thanks again for the crash report. It helps a lot to understand where was the problem. I make a fix for this. It will be available on Sunday.

exelban, 2020-11-25:
I have made a build with this fix. You can use it till the new release.
[Stats.dmg.zip](https://github.com/exelban/stats/files/5597262/Stats.dmg.zip)

crownzdp, 2020-11-25:
thanks,I'll try it tomorrow

exelban, 2020-11-29:
Fixed in the [last](https://github.com/exelban/stats/releases/tag/v2.4.5) release.

closed 2020-11-29 by exelban