exelban/stats #191: More GPU info

sukmaraga, 2020-11-25:
THANK YOU for making this wonderfull app!
this is the only app that show my gpu (rx560) temp on my hackintosh (even istat cant)

1)
i ussually monitor gpu stats with terminal. i wonder if you planning to add this to the app
[image]

`while sleep 1; do clear;ioreg -l |grep \"PerformanceStatistics\" | cut -d '{' -f 2 | tr '|' ',' | tr -d '}' | tr ',' '\n'|grep 'Temp\|Fan\|Clock'; done`

2) sensors popup is too big, and cant fully visible with my 720 pixel height tv. maybe temp and power should be separated ? 

thanks again!

exelban, 2020-11-25:
Hi. Thanks for your feedback.

1. Yes, it's possible. But there a few problems with this.
First of all, I need some layout for this. The second problem is the GPU reader. There is a lot of GPUs which not provide these statistics. And when users will see `Core clock: undefined` they will be opening issues with this. And saying that this is a bug. So to implement these statistics I need to have UI. And this UI must be built in a little bit modular way. There must be 2 parts for each GPU: charts with utilization and temperature and additional statistics. 
TR.DL: I need some UI that visualize the statistics which are available for GPU. And it must look "natural" without utilization, temperature, core clock, etc data.

2. The sensor popup view must fit the display. And scroll must be available. Could you please post the screenshot if it's not working?

sukmaraga, 2020-11-26:
1.
yes, it should be modular or something, as those stats maybe only come with dedicated GPU like on macpro, as i tried on my macmini with iGPU dont show anything. maybe adding toggle in GPU setting as 'experimental', so user can turn in on or off.

2. 
i latter realize the popup the fit with the screen! its just covered under dock :D 
[image]

exelban, 2020-11-26:
1. The problem with missing stats is released not only to integrated GPUs or only some vendor. Even the same GPU on different macs have not the same stats from time to time. So just to conclude this thread. I need some UI for this.

2. )

closed 2021-01-10 by exelban