exelban/stats #192: reporting a bug

mutawa19, 2020-11-25:
Since I installed Stats app on my laptop, I have noticed that it drains the battery while mac is on sleep mode.
I have checked the activity monitor to see if there is any app preventing sleep but found none. I am sure that there is a bug with Stats app because I checked the battery level after a night with Stats enabled on, and another night with the app disabled. When, the app is disabled, the battery level dose not change when I check it the next day.

exelban, 2020-11-25:
Hi. Could you please share the energy consumption of Stats?

mutawa19, 2020-11-25:
[image]

Hi, I am not shore how to find the energy consumption of Stats. Did you mean the energy impact in the activity monitor?

exelban, 2020-11-25:
Yeap. Thanks. It looks like it has a low energy impact on mac.

exelban, 2020-11-25:
There is almost no option that the app will be working in sleep mode.
PS: are you sure that mac was in sleep mode?

mutawa19, 2020-11-25:
Yes, I have that set to go to sleep automatically after 10 minutes of inactivity. In fact, I manually put it to sleep last night.

exelban, 2020-11-25:
Based on the energy impact there is no option that this app has a big impact on power consumption. If the mac sleep, Stats app does not work. I suppose in the sleep mode every app is frozen.

mutawa19, 2020-11-25:
I thin it is impacting the battery. Here is what I will do to confirm that.
 I will close all apps except Stats and put the laptop on sleep mode. I will check it after one hour and take screenshot of the activity monitor. I will then repeat the test again with the Stats closed and show you a screenshot of the activity monitor after another hour of sleep.

I hope that helps in convincing you that there is a bug in Stats app.

exelban, 2020-11-25:
)
You don't need to convince me that there is a bug in Stats. If there is a problem in the Stats you don't need to put the mac in sleep mode. If the app drains the battery in sleep mode. It means that the mac is not in this mode.

mutawa19, 2020-11-25:
That is what I thought and mentioned in my first comment.

So are you now saying that **Stats is** preventing sleep even though activity monitor showing otherwise?

exelban, 2020-11-25:
I'm just saying that if mac is on sleep mode. The app does not work.
Stats are not using any exceptions that could wake up mac. But even if it could, you will see that your mac is not in sleep mode.

exelban, 2020-11-29:
Hi. Any updates on this?

wangzhaoming, 2020-11-30:
I've got the same problem. 
I put my mac on sleep mode without quiting Stats and it consumes 25% of battery over night. And my mac consumed only 3% of battery when I quit Stats.

According to the `pmset -g log`, I think Stats did not prevent sleep.

In addition, I'm using version 2.4.4 and the energy impact is low according to the Activity Monitor.

I suspect it has something to do with the Intel Power Gadget. It's just my guess, there is no basis.

wangzhaoming, 2020-12-01:
I uninstalled Intel Power Gadget and the battery consumption seems to be normal last night.
@mutawa19 Maybe you can do some more tests if you installed the Intel Power Gadget too.

mutawa19, 2020-12-01:
> Hi. Any updates on this?

I am still having the same issue.

mutawa19, 2020-12-01:
> I uninstalled Intel Power Gadget and the battery consumption seems to be normal last night.
> @mutawa19 Maybe you can do some more tests if you installed the Intel Power Gadget too.

Hi @wangzhaoming, Thank you for the suggestion.  
I have the intel Power Gadget installed on my mac but I rarely use it. I am not sure if Stats app actually depends on other apps to function. I will try your suggestion and let you know in a couple of days. 

Thanks.

wangzhaoming, 2020-12-01:
Stats reads CPU frequency by the Intel Power Gadget.
After removing the Intel Power Gadget, the CPU frequency module will disappear. Need to search Intel Power Gadget from Finder to remove it completely.
[image]

exelban, 2020-12-01:
Hi. Thanks for the suggestion. But I'm not sure that IPG makes a problem. Because.

In the app, there are few types of readers (a function which read some values: cpu usage, ram utilization, cpu frequency, etc). To make the app more power-efficient. I split readers into 3 types: 
- required: runs all time
- optional: could be replaced by required and not running by default
- popup: runs only when popup view is opened and stopped if not

And the CPU frequency read the value in `popup` reader. Which are running only when the popup is opened. So, if you not leave the CPU popup opened all the time. The app is not reading the frequency from IPG.

exelban, 2020-12-01:
https://github.com/exelban/stats/blob/950b787c73a34ef98b1e544e2e78a5abddcbf865/ModuleKit/module.swift#L303-L316

exelban, 2020-12-01:
Or wait. Maybe it could be. Because I rewrite the CPU frequency module and could miss set this reader as `popup`(

exelban, 2020-12-01:
Yes, it could be the frequency reader from IPG. Because I rewrite the CPU frequency reader in the last release (to make the app compatible with Apple Silicon). And I do not set this reader as `popup`. `popup` means that the value will be reading only when the popup is opened.

I will fix it today, and make a test build for you.

exelban, 2020-12-01:
I will decrease CPU consumption (at 5-8%). But it not resolve the bug with sleep mode. Because even IPG does not work in sleep mode.

exelban, 2020-12-01:
[Stats.dmg.zip](https://github.com/exelban/stats/files/5624063/Stats.dmg.zip)

mutawa19, 2020-12-02:
> Stats reads CPU frequency by the Intel Power Gadget.
> After removing the Intel Power Gadget, the CPU frequency module will disappear. Need to search Intel Power Gadget from Finder to remove it completely.
> [image]

I tried your suggestion but I still have the same issue even after uninstalling Intel Power Gadget. I removed IPG using app cleaner which supposed to remove completely, however, the Stats is still showing the CPU frequency module on my mac. I am not sure why that is. 

[image]

macOS version: Mojave 10.14.6 
Stats version: 2.4.6

wangzhaoming, 2020-12-02:
> I tried your suggestion but I still have the same issue even after uninstalling Intel Power Gadget. I removed IPG using app cleaner which supposed to remove completely, however, the Stats is still showing the CPU frequency module on my mac. I am not sure why that is.

Need to remove `/Library/Frameworks/IntelPowerGadget.framework` @mutawa19 .
Search `intelpowergadget ` using the Finder will find it.

mutawa19, 2020-12-02:
Thanks @wangzhaoming. I removed the IPG.framework and now the CPU frequency disappeared after restarting Stats. I think that might fix the battery draining issue. I'll check the battery level tomorrow and let you know.

exelban, 2020-12-02:
Hi.

I have made some tests. I charge my MacBook to 100% and leave it till discharge. I repeat this a few times. Once with running Stats, and next without. Unfortunately, I have Big Sur on MacBook, so it leaves only 2h(
This time does not change if the Stats was running or not. 

I tried a few times to check a sleep mode. After night the battery percentage was the same with Stats and without. 
I'm noticing from time to time, that when I close my MacBook, or event click to sleep. It does not go into sleep mode. Because an hour later I hear a Slack notification)

I really understand the problem. Because if in normal mode app has not big power consumption. And even, IF in some way, the app still works in sleep mode (what are almost impossible). The consumption will be the same as in wake mode.

I cannot do more at this moment.

PS: IPG is only one way to have a CPU frequency at intel based mac. You will not see CPU frequency if you don't have installed IntelPowerGadget framework.

mutawa19, 2020-12-03:
The battery was at 94% this morning which is much better than before when I had IPG.frameworks. 
Before removing the IGP, the battery used to be at 40 - 50% after a night with Stats running in sleep mode.

wangzhaoming, 2020-12-03:
> [Stats.dmg.zip](https://github.com/exelban/stats/files/5624063/Stats.dmg.zip)

I reinstalled the IPG and tested this version of Stats last night. The battery consumption is low.
It seems to be working for me.

wangzhaoming, 2020-12-03:
Can you add a CPU frequency option? If I turn it off, the Stats won't initialize the IPG framework even if IPG is installed on my mac.

mutawa19, 2020-12-03:
> Can you add a CPU frequency option? If I turn it off, the Stats won't initialize the IPG framework even if IPG is installed on my mac.

No, CPU frequency is not displayed.
I have just installed Stats version that you provided here and CPU frequency is still not showing. I think because I already uninstalled IPG. 

Do you want me to install IPG and test again?

wangzhaoming, 2020-12-03:
> No, CPU frequency is not displayed.
> I have just installed Stats version that you provided here and CPU frequency is still not showing. I think because I already uninstalled IPG.
>
> Do you want me to [...]

Oh, I'm sorry I made you misunderstood. I'm a user of Stats, not developer 😄 .  I'm wondering if @exelban can add a CPU frequency for Stats.

closed 2020-12-06 by exelban

mh-saeed, 2025-04-13:
Is this normal?

[image]