exelban/stats #193: Feature Request: Add CPU throttling/speed limit info

ikester, 2020-11-25:
Hi. I'm really enjoying Stats. Great work!

I recently discovered a project called Hot: https://github.com/macmade/Hot

It shows the CPU temperature and indicates if the CPU is being throttled. It's MIT licensed and I think it would be great if that info could be added to Stats, either under the Sensors section, or perhaps as its own category.

Hopefully you'll find this useful.

exelban, 2020-11-25:
Hi.

Thanks. I take a look at this app. But cannot really understand what it does.
Showing the utilization and the temperature of the CPU?

ikester, 2020-11-25:
Not the utilization. I believe it's showing how much the CPU is being throttled. 100% meaning not throttled at all. But I just discovered it, so I could be wrong about how to interpret it.

exelban, 2020-11-25:
There is an option to colorize the value of the CPU usage (based on utilization). I make some tests with this app. And it always shows 100% for me. Under load and without.

ikester, 2020-11-25:
I don't think this is just for when the CPU is under load. I think the CPU has to be throttled down due to thermal issues. If your Mac has good fans, it may never get throttled.

exelban, 2020-11-25:
Ohh, I understand now. So it just shows the potential throttling based on the CPU temperature.

I take a look at the code. It using the internal tool `pmset` to fetch `CPU_Scheduler_Limit` and `CPU_Speed_Limit`.

I'm not sure that it's useful information.

exelban, 2020-11-25:
I suppose it could be added as additional information in the popup view. If someone really needs this information.

exelban, 2020-11-29:
Reopen this issue if someone will need this information in the popup view.

closed 2020-11-29 by exelban

ryadom, 2021-06-21:
I would like to see this sensor in the app.
My macbook throttles when it's 30+°C and sunny outside :-)