exelban/stats #194: The icons on the menu bar are too widely spaced.

kangkangk, 2020-11-26:
Hi,I am very grateful to you for developing such a great application that keeps me informed about the working status of my computer. However, the spacing of the icons on the menu bar is not consistent with other applications. I would like to add an option to adjust the distance between the icons at will to make them look better!

exelban, 2020-11-26:
Hi.
Are you using macOS Big Sur or Catalina?

kangkangk, 2020-11-26:
> Hi.
> Are you using macOS Big Sur or Catalina?

I am using Big Sur

exelban, 2020-11-26:
So please take a look at [this](https://github.com/exelban/stats/issues/164) issue.

kangkangk, 2020-11-26:
> So please take a look at [this](https://github.com/exelban/stats/issues/164) issue.
I see. Thank you.

closed 2020-11-26 by exelban