exelban/stats #195: Feature request: Hide battery text when "n/a"

igorkulman, 2020-11-26:
**Describe the bug**
When the text next to battery shows as "n/a" it does not look good, it looks some kind of error. I want the battery widget to show the time remaining, but only when the time remaining is computed. For example when plugged in it always shows "n/a" which does not look OK to me.

**Details:**
 - MacOS 10.15.7
 - Application version 2.4.4

exelban, 2020-11-26:
Hi. 
There is an option to hide additional text when the battery is charged.
There was already a [discussion](https://github.com/exelban/stats/issues/138) about n/a.

igorkulman, 2020-11-26:
Looking at the discussion, yes it is the same problem. But I think the solution is a bit different. The solution is juts so "n/a" is not shown when fully charged and plugged in. There are other cases, for example when you unplug the computer and it is not fully changed you get "n/a" for some time until macOS computes the remaining battery time.

exelban, 2020-11-26:
Yeap, this moment was discussed. And I decided to leave `n/a` till a better solution.
Do you propose to hide time (`n/a`) when it's not available?

igorkulman, 2020-11-26:
Yes, my idea is  if text is "n/a" then hide it.

exelban, 2020-11-26:
Hmm, I could add this.
I don't like only one moment. If two rows (percentage + time) option is selected. And time will be 0 (`n/a`). The widget will show only 1 row, percentage. Abd 

I would prefer to give some information about time is calculating.

exelban, 2021-03-28:
Hi. I have no better solution than have `n/a`. The issue will be closed next week. If there will be no good ideas.

closed 2021-04-04 by exelban