exelban/stats #197: Feature Request: Fan speed & temp sensor stat in one icon

yokonrad, 2020-11-28:
It would be great to have an option to display fan speed and temperature sensor (cpu) stats in one icon similar to smcfancontrol. Of course without control of fans. If they are separated they consume much space which could be saved when merged into the one. Additionaly smcfancontrol is not updated to operate on big sur on 13 inch macbook pro retina. There is an issue at their github but project seems to be dead. Picture attached how it looks with smcfancontrol. I would totally drop of using their app if you could add this function to your app :)

[image]

exelban, 2020-11-28:
Hi. Thanks for the feedback.

You can't do it with one icon. But you can combine two: sensors and fans. The difference will be in the layout. 
But it allows you to specify what exactly temperature you want to see. 

[image]

There will be no option to show temperature and fan in one icon. Because it's a separate modules.

yokonrad, 2020-11-28:
I know that i can combine two separated icons with temperatures and fans. But the macbook pro 13' (late 2013) have only one fan and making icon with one stat looks bad when you compare to the others. Just look on the sshots. There is also smcfancontrol not updated on the sshot about which i mentioned earlier

[image]
[image]

exelban, 2020-11-28:
Ok, I see.
I think I can add fans to the sensors module. And then you could select the temperature sensors and fan in the sensor module.

It will look like this.
[image]

yokonrad, 2020-11-28:
Great, thanks! :) I think temperature should be aligned to the left just like fan sensor because its a value. When its aligned to the right it looks like the name of the unit (like RPM). Or maybe centered with space between value and name of the unit? (if its even possible)

exelban, 2020-11-28:
For now, there is no option to select an alignment for this. But I think it could be added in the future.

exelban, 2020-11-29:
Available in the [new release](https://github.com/exelban/stats/releases/tag/v2.4.5).

closed 2020-11-29 by exelban

yokonrad, 2020-11-30:
> There is no option to select an alignment for this, for now. But I think it could be added in the future.

Would be awesome if you can align it in the future. Anyway i found two bugs:
1. App is showing wrong model of my mbp. I have late 2013 and app is showing mid 2014
[image]
[image]

2. Different fan "names" in modules Fans and Sensors. "Right side" in Fans module and "Fan #0" in Sensors module
[image]
[image]

P.S. Does the width of the CPU menu bar icon have to be that big? Even with 100% of usage it could be a little smaller. Same question about Sensors menu bar icon. Look at a blank space between icons.
[image]
[image]

Others menu bar icons are good.
[image]

exelban, 2020-11-30:
1. I suppose you have `MacBookPro11,1`. According to [this](MacBookPro11,1) there were 2 models with the same id: late 2013 and mid 2014. I can identify the mac only by this id.
2. It's normal. In the sensors, there are no fans names.
3. Take a look [here](https://github.com/exelban/stats/issues/164).

yokonrad, 2020-12-04:
> * I suppose you have `MacBookPro11,1`. According to [this](MacBookPro11,1) there were 2 models with the same id: late 2013 and mid 2014. I can identify the mac only by this id.
> * It's normal. In the sensors, there are no fans names.
> * Take a look [here](https://github.com/exelban/stats/issues/164).

Yes, you are right ;) Thanks for answers.