exelban/stats #199: CPU Frequency stops updating after a MSR is written

JJGadgets, 2020-11-29:
I use VoltageShift and its MSR write feature to undervolt and underclock my MacBook Pro when on battery for much better thermals and battery life (I do this on all laptops). stats is the first free tool I've used that allows CPU frequency monitoring. However, after writing to the MSR 0x1AD (which is the max frequency MSR iirc, wrote my script a while back), the CPU frequency monitoring seems to stop working. Is it possible to fix this?

exelban, 2020-11-29:
Hi. Thanks for reporting.

App using IntelPowerGadget to read the frequency. I never use undervolting or underclocking on mac. And I don't know the impact on the IPG. Do you maybe have an IntelPowerGadet? To check if it's showing the values after undervolting or underclocking?

JJGadgets, 2020-11-29:
Intel Power Gadget is updating properly, but stats is not.
Well actually it used to, today it seems to be fine and I've messed around with that MSR a few times today.
I'll let you know after a few days, because I think opening Intel Power Gadget after applying a new MSR might fix it (that's what I did subconsciously today).

JJGadgets, 2020-11-29:
Actually, speaking of frequency reporting, could you add an option to display the frequency in the text field and the CPU usage in the graph, and vice versa, and maybe an option to show 2 of them in one menubar item (2 graphs?)

exelban, 2020-11-29:
I will recheck this one. Because it's possible that after MSR intel closing a "channel". And maybe it requires reconnection.

There is no option to have a few widgets for one module. I'm have already thinking about this one. But it requires rewriting a big part of the app. And now I don't have enough time for this. In the future, it will be possible to have a few widgets for one module.

exelban, 2020-11-30:
I think I figured out where is the problem. 

To read the frequency I must pass 2 points in time to IPG. If IPG throws the error on the reading frequency it returns from read function. And it brokes the first point. Because it will be never updated. The one way to fix it, restart the app.
I think it's also the reason why it stops working after MSR.

The fix will be available in the next release (on Sunday).

JJGadgets, 2020-12-01:
Alright, sweet, will let you know if the new release fixes it when it's released.

closed 2020-12-06 by exelban