exelban/stats #201: Segmentation fault on Mojave

chenguokai, 2020-11-30:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: Macbook Pro 2017
 - MacOS 10.14.6
 - Application version 2.4.5

exelban, 2020-11-30:
Hi. Could you please post the crash report?

chenguokai, 2020-11-30:
> Hi. Could you please post the crash report?

I am not quite sure about what you referred to. But if you refer to the log in a window that usually pops up after an app crash, the answer is sadly no. If being launched from GUI, it just crashes without any feedback. I figured out that it is a segmentation fault by launching it using terminal.

exelban, 2020-11-30:
You can find a crash report in `Applications -> Utilities -> Console`. Find the `Crash Reports` tab. There you must see a row called `Stats`.

chenguokai, 2020-12-01:
[Stats_2020-12-01-111236_XIM-MacBook.txt](https://github.com/exelban/stats/files/5620094/Stats_2020-12-01-111236_XIM-MacBook.txt)

Hope this would help :-)

exelban, 2020-12-01:
Thanks. It looks like [this](https://github.com/exelban/stats/issues/202).
It's already fixed and a [new](https://github.com/exelban/stats/releases/tag/v2.4.6) version released.

closed 2020-12-01 by chenguokai