exelban/stats #202: Crashed on v2.4.5

cvluca, 2020-11-30:
**Describe the bug**
Issue occurrs after updated to v2.4.5. It doesn't show up on menu bar, and it just crashed at start.

**Details:**
 - Device: [iMac (Retina 4K, 21.5-inch, 2019)]
 - MacOS [10.15.7]
 - Application version [2.4.5]

Here is the crash report:
```
Process:               softwareupdated [1866]
Path:                  /System/Library/CoreServices/Software Update.app/Contents/Resources/softwareupdated
Identifier:            softwareupdated
Version:               1293.2.1
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
Responsible:           softwareupdated [1866]
User ID:               200

Date/Time:             2020-11-30 22:54:29.681 +0800
OS Version:            Mac OS X 10.15.7 (19H15)
Report Version:        12
Anonymous UUID:        FAFE0E59-61FE-10B6-994C-14AC07C61DFA

Time Awake Since Boot: 160 seconds

System Integrity Protection: enabled

Crashed Thread:        2  Dispatch queue: NSOperationQueue 0x7fbd06e09230 (QOS: UNSPECIFIED)

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1866]

Application Specific Information:
dyld3 mode
rdar://problem/28724618 Couldn't connect to launchservicesd.

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6e783dfa mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff6e784170 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff345a4ef5 __CFRunLoopServiceMachPort + 247
3   com.apple.CoreFoundation      	0x00007fff345a39c2 __CFRunLoopRun + 1319
4   com.apple.CoreFoundation      	0x00007fff345a2e3e CFRunLoopRunSpecific + 462
5   com.apple.CoreFoundation      	0x00007fff3462b489 CFRunLoopRun + 40
6   softwareupdated               	0x0000000105fcd9f7 0x105fcc000 + 6647
7   libdyld.dylib                 	0x00007fff6e642cc9 start + 1

Thread 1:: Dispatch queue: com.apple.SoftwareUpdate.SUScanController.ScanRefresh
0   libsystem_kernel.dylib        	0x00007fff6e783e4e semaphore_timedwait_trap + 10
1   libdispatch.dylib             	0x00007fff6e5e9b6b _dispatch_sema4_timedwait + 76
2   libdispatch.dylib             	0x00007fff6e5e9f97 _dispatch_semaphore_wait_slow + 58
3   com.apple.SoftwareUpdate.framework	0x00007fff640bbcea -[SUScan _scanMain:] + 1009
4   com.apple.SoftwareUpdate.framework	0x00007fff640bb8f2 -[SUScan performSynchronousScan] + 88
5   com.apple.SoftwareUpdate.framework	0x00007fff64119a3f __298-[SUScanController refreshAvailableProductsInForeground:limitedToChanged:evenIfConfigurationChanged:initiatedByDeviceConnection:limitedToProductKeys:limitedToProductTypes:forCurrentConfiguration:distributionEnvironment:distributionEvalutionMetainfo:installedPrinters:preferredLocalizations:finish:]_block_invoke_2 + 520
6   libdispatch.dylib             	0x00007fff6e5e86c4 _dispatch_call_block_and_release + 12
7   libdispatch.dylib             	0x00007fff6e5e9658 _dispatch_client_callout + 8
8   libdispatch.dylib             	0x00007fff6e5eec44 _dispatch_lane_serial_drain + 597
9   libdispatch.dylib             	0x00007fff6e5ef5d6 _dispatch_lane_invoke + 363
10  libdispatch.dylib             	0x00007fff6e5f8c09 _dispatch_workloop_worker_thread + 596
11  libsystem_pthread.dylib       	0x00007fff6e843a3d _pthread_wqthread + 290
12  libsystem_pthread.dylib       	0x00007fff6e842b77 start_wqthread + 15

Thread 2 Crashed:: Dispatch queue: NSOperationQueue 0x7fbd06e09230 (QOS: UNSPECIFIED)
0   libobjc.A.dylib               	0x00007fff6d48fd24 objc_autoreleasePoolPop + 25
1   com.apple.CoreFoundation      	0x00007fff34562a85 _CFAutoreleasePoolPop + 22
[... 599 lines cut ...]
JS JIT generated code              1.0G        3 
JS VM Isolated Heap                 44K        2 
Kernel Alloc Once                    8K        1 
MALLOC                           259.3M       85 
MALLOC guard page                   72K       15 
MALLOC_MEDIUM (reserved)           1.4G       12         reserved VM address space (unallocated)
STACK GUARD                       56.1M       19 
Stack                             17.1M       19 
VM_ALLOCATE                        556K        5 
WebKit Malloc                     21.7M       34 
__DATA                            38.8M      403 
__DATA_CONST                        53K        5 
__FONT_DATA                          4K        1 
__LINKEDIT                       388.9M        6 
__OBJC_RO                         32.3M        1 
__OBJC_RW                         1908K        2 
__TEXT                           305.6M      393 
__UNICODE                          564K        1 
libnetwork                        1664K       24 
mapped file                       84.5M       12 
shared memory                       44K        8 
===========                     =======  ======= 
TOTAL                              3.6G     1056 
TOTAL, minus reserved VM space     2.2G     1056 
```

exelban, 2020-11-30:
Hi. Thanks for reporting.
But could you please post a crash report of the Stats app?)

cvluca, 2020-11-30:
> Hi. Thanks for reporting.
> But could you please post a crash report of the Stats app?)

Sorry, I copied the wrong one. Just updated.

exelban, 2020-11-30:
Do you have installed IntelPowerGadget?

cvluca, 2020-11-30:
> Do you have installed IntelPowerGadget?

Yes, I do

exelban, 2020-11-30:
Do you maybe know what version do you have?

cvluca, 2020-11-30:
> Do you maybe know what version you have?

3.5.5

exelban, 2020-11-30:
Hmm. It's crashing because it cannot call a function from IntelPowerGadget. I can't install `v3.5.5`(
I will try to install this version manually and wrap this bug. Could I ask you to update an IntelPowerGadget? It will fix the crash.

exelban, 2020-11-30:
Or I can post a version that catches this bug. But you will not see a CPU frequency in the popup view.

cvluca, 2020-11-30:
> Hmm. It's crashing because it cannot call a function from IntelPowerGadget. I can't install `v3.5.5`(
> I will try to install this version manually and wrap this bug. Could I ask you to [...]

works now after updated to the last version of IntelPowerGadget

exelban, 2020-11-30:
Thank you for reporting. It's a critical bug. I will release a new version with this fix today.

exelban, 2020-11-30:
The [new](https://github.com/exelban/stats/releases/tag/v2.4.6) version with this fix released. Thanks.

closed 2020-11-30 by exelban